Ultimaker/Cura #15286: Failed to Slice

jchihak-stlcc, 2023-04-18:
### Application Version

5.3.0

### Platform

Apple macOS 12.6.1

### Printer

Ultimaker 2+ and Ultimaker 3 Extended

### Reproduction steps

1. Imported 3MF file into Cura
2. Applied desired print settings 
3. clicked Slice button

### Actual results

Slicing begins and before reaching ~10% fails with error Slicing failed
Slicing failed with an unexpected error. Please consider reporting a big on our issue tracker.

### Expected results

Model slices successfully and then saved to disk or removable media.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11265012/cura.log)
[image]
[UM3E_3-D printing.3mf.zip](https://github.com/Ultimaker/Cura/files/11265018/UM3E_3-D.printing.3mf.zip)

GregValiant, 2023-04-18:
Thanks for the report and the files.
From you log file:
"Backend exited abnormally with return code -11!"
Someone from the Cura Team will know what that means.  I'll mark this as a duplicate report and a slicing error.

I moved the model and then rotated it 90° CCW and it sliced.  This bug showed up in 5.2 and it has been worked on but it obviously isn't gone yet.
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic