Ultimaker/Cura #15288: cannot open a file on iMac 2011 on sys 11.7.6. i cannot open from file open or recent or by dragging

ravgu, 2023-04-18:
### Application Version

5.3.0

### Platform

Mac

### Printer

Creality enders3

### Reproduction steps

I tried to open a .stl file to slice. 
In does not open from file open
or from recent
or by dragging
but it opens in Cura3

### Actual results

no results to open a file
All preferences and market place opening works

### Expected results

file should open to manipulate and slice

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11265792/cura.log)

sh3t0r, 2023-04-26:
I have the same issues on macOS 11.7.6.
Cura 5.2.1 works without a problem.

MariMakes, 2023-05-05:
Hey @ravgu (and @sh3t0r)

Thanks for your report 👍
Your Mac versions should be supported so that's odd 🤔 

This reminds me of a recent fix that might fix this issue. 
It's a bit hard for us to test, but we have a nightly build available.

Can you please try it out and let us know if it resolves your issue?
You can find the download links and the discussion here: https://github.com/Ultimaker/Cura/discussions/15098

sh3t0r, 2023-05-05:
I’d love to. Is there a nightly build for macOS?

Mit freundlichem Gruß,

Christoph Ohlrogge

ravgu, 2023-05-06:
What I’ve found it does work but takes a long time to open. If I give an open command and leave it for an hour it will open and subsequent opens happen quickly