Ultimaker/Cura #15289: Pause at height resume z-height too high

ThatsNotAMoon, 2023-04-18:
### Application Version

5.3.0

### Platform

windows 11

### Printer

Ultimaker S5

### Reproduction steps

I sliced with the post-process pause at the layer. Once the print was resumed the z-height was about an inch higher than the model

### Actual results

Once the print was resumed the z-height was about an inch higher than the model

### Expected results

The resume height should be at the model +1 layer to start the next layer

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://drive.google.com/drive/folders/1AzYoSVv3bIUfRnGviyEza5WUjfSXq1RV?usp=share_link

ThatsNotAMoon, 2023-04-28:
Thanks for your reply. The S5 does not allow you to set a disarm timer it seems so I have no control over that. 

While watching the pause happen by setting the first layer to pause, I have found that when the pause happens, the z motor itself seems to skip a few steps before lowering to the base which seems to be the cause of the lost position. My rods and Z-screw are in good condition so I don't see a reason for this issue and my motor seems to be fine for everything else so I wouldn't think I've got an issue with the hardware either.

ThatsNotAMoon, 2023-04-28:
A dumb mistake! I installed the latest cura but it didn't uninstall the old cura, which is what I was still slicing on, which had an issue with the pause feature for the griffin firmware haha. Bone head mistake

closed 2023-04-28 by ThatsNotAMoon

GregValiant, 2023-04-28:
Maybe, but it takes a brave person to admit something like that.  Thanks for closing this.