Ultimaker/Cura #15290: Z-seam Misalignment

cetemadi, 2023-04-19:
### Application Version

5.3.0

### Platform

macOS 13 and Windows 10

### Printer

Creality Ender 5

### Reproduction steps

1. Set Z Seam Alignment to User or Sharpest Corner
2. Select None as Seam Corner Preference.
3. Results are very different compared to version 4.2, seam is not aligned and seems random.
[CE5_Caps.3mf.zip](https://github.com/Ultimaker/Cura/files/11267549/CE5_Caps.3mf.zip)

Cura 5.3
[image]

Cura 4.2
[image]

### Actual results

Results are very different compare to version 4.2, seam is not aligned and seems random.

### Expected results

Z seam should be aligned at the corner or user coordinates. Instead the Z seam is misaligned and looks random

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE5_Caps.zip](https://github.com/Ultimaker/Cura/files/11267678/CE5_Caps.zip)

Vandresc, 2023-04-28:
Hi @cetemadi , thank you for your feedback. We have some tickets on the backlog related to the seam so I will add your scenario to our internal ticket, devs see https://ultimaker.atlassian.net/browse/CURA-10219.

eugenmeltis3pg, 2023-08-20:
Is there any news on this?

Erosion139, 2024-10-08:
Bump because its 2024 and this issue is still persisting for me. 
[image]

This is with seam set to sharpest with no corner preference.