Ultimaker/Cura #15295: slicer failed

XtCsoldier, 2023-04-19:
### Application Version

latest

### Platform

MAC

### Printer

Dremel 3D40

### Reproduction steps

Everytime I slice any model it states it's unable to slice

### Actual results

Nothing happens I cannot create a file for the printer

### Expected results

Slice properly and create a .gcode

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Lighsaber.silver.3mf.zip](https://github.com/Ultimaker/Cura/files/11277174/Lighsaber.silver.3mf.zip)

GregValiant, 2023-04-20:
Thanks for the report and the files.
I didn't have any trouble slicing your project file (I'm on a Windows 11 system).  If this was a widespread problem, nobody would use Cura so it may be specific to your system.  What is the name of the operating system you are using, and if you know the name of the processor that is in your computer passing that along might help to troubleshoot.
[image]

I'll leave the bug label on this and maybe another MAC user will have a suggestion.  It could be that another piece of software is interfering or you may have to do a re-install if you are having trouble slicing anything at all.

Just as a note - when you have so many models on the build plate at once, a failure of one can cascade into the others.  This is also a 24 hour print.  If you split off the two big models you will have two 12hr prints.  If nothing else that should relieve your anxiety level.

XtCsoldier, 2023-04-20:
This was the original I tried to slice, but it won't allow me to slice even a single one of the parts. I will try a reinstall. I am using a MAC with an M1 Pro chip and operating system Ventura 13.3.1

XtCsoldier, 2023-04-20:
After reinstalling it will not allow me to slice anything that has even a moderate amount of complexity. Models I have literally seen printed in the printer I am trying to send the file to. Attempting to run a Windows 11 off of a VPN to see if I can recreate your success