Ultimaker/Cura #15296: Staggered layer pattern

wheresmc, 2023-04-19:
### Is your feature request related to a problem?

Firstly, my apologies in advance if this is a non-issue with a solution that I am not aware of.

I am using the concentric pattern for my current clear PETG print (top, bottom and infill) and while checking out the preview I that the layout with the directly stacked pattern might not be the most optimal and that alternating the pattern might be advantageous.

### Describe the solution you'd like

The general idea is to print the bead on the seams of the previous layer - theoretically, this should sandwich each (internal) bead between 6 other beads (instead of 4) and improve inter-bead adhesion of the layer below by "brigading" and bonding the two accent beads together and applying addition heat directly on the seam.

This feature could be a selectable option within the "infill" and/or "top/bottom" menu.

### Describe alternatives you've considered

While not a problem, perhaps the feature might improve print strength and/or quality via improved inter-bead and/or inter-layer adhesion.

### Affected users and/or printers

All users and all printers, but most applicable when using 100% infill.

Vandresc, 2023-04-20:
Hi @wheresmc, welcome to the UltiMaker Cura Github 🚀 
I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞

GregValiant, 2023-04-20:
I like this idea.  This would address the "philosophical problem" I have with Concentric.
Given any shape with vertical walls, each "concentric" pattern is immediately above the previous pattern.  It is the same as having all "Lines" or "ZigZag" lines running in the same direction.  I don't know that it has much of an effect on the strength, but it seems like it would.

wheresmc, 2023-04-21:
Hi Greg,

Thanks for adding your support!

If ever makes it into even "experimental mode" will be interesting to see
if it actually makes any difference - especially when taking additional
slicing and printing time into account.

Best,

MC

smartavionics, 2023-04-22:
It's an interesting idea, I did a little experiment with my Cura 4 based fork and simply shrunk the skin area by 1/2 a skin line width on even numbered layers and it does this...

[image]

[image]

You can see on layer 2 the outermost skin line is 1/2 the normal width and so the skin lines inside are shifted. That thin outermost skin line is actually produced by the fill gaps between walls setting (so it needs to be enabled to avoid having a gap around the outside of the skin).

Anyway, I'll put this into my next release (currently not scheduled but will be along some time).