Ultimaker/Cura #15297: 5.3 Small Feature Walls, inner layer not slowed

reid-p, 2023-04-20:
### Application Version

5.3

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Slicing with the same profile in 5.3 vs 5.2.2, small feature wall speed is different.

### Actual results

Here's the 5.2.2 speeds:
[image]

Here's the 5.3 speeds:
[image]

### Expected results

The inner wall speed for small features should be reduced by set amount.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

MariMakes, 2023-05-05:
Hey @reid-p,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

What you are seeing seems to be introduced on purpose in 5.3, You can read more about it here: https://github.com/Ultimaker/CuraEngine/pull/1777
_The logic for slowing down "small features" should only apply to types of extrusion paths for which it makes sense: outer walls, support outer walls, and maybe a few others. Not for extrusion paths that are completely enclosed within and constrained by another outer structure where printing them slowly will not contribute to dimensional accuracy and surface quality of the part_

Can you help me understand why the old behavior was better for you or why the new behavior is impacting you?
For example, if you see a difference in your printed part quality and have a picture to show it I can raise it with the team. 
Thank you 🙏

reid-p, 2023-05-09:
G'day @MariMakes.

I'm was seeing problems with the small holes. Things weren't quite right, which I didn't see with the old version.
Tweaked a couple of things and it's good enough now.

It may have been ordering of the wall print order. 

The logic for the change in #1777 is that the inner walls are constrained.
But that'd depend on the print order right?.

Perhaps if the inner walls are printed first than the speed should be reduced using the old logic?

Thisismydigitalself, 2023-09-30:
I as well using a lot this small features feature and love it but the fact that Cura does not show me the actual speed of those makes it hard if not impossible to know that the setting are correct  and it goes without saying that the speed preview does not corresponds to the actual speed in which these will get printed.

Cura 5.4.0/Wintel64