Ultimaker/Cura #15302: Adaptive Layers Topography Size not working as expected

upatamby, 2023-04-20:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Ultimaker 3E

### Reproduction steps

1. load the Cura project file (see uploaded zip-file)
2. open preview
3. look at the bottomside of the lowest winding of the worm (see image) 
[image]

### Actual results

For this part of the model the Adaptive layering (in combination with the Adaptive Layers Topography Size-parameter) is not working as expected. 

### Expected results

The horizontal distance between two adjacent layers should not exceed 0,1mm. This works fine for all layers, except for those few layers at the bottom of the worm.
If I change the Adaptive Layers Variation Step Size from 0.01 to 0.05 the result is better, but still not exactly as it should be. It seems that Cura doesn't anticipate to the change of layer thickness in time, but only for this part of the model. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[wormas.zip](https://github.com/Ultimaker/Cura/files/11290174/wormas.zip)

MariMakes, 2023-05-05:
Hey @upatamby,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

I'm not sure if I understand correctly you expect the first "big overhang layer" to be thinner than it is, right? 
[image]

Your issue reminds me of https://github.com/Ultimaker/Cura/issues/11160. [There is a really good explanation](https://github.com/Ultimaker/Cura/issues/11160#issuecomment-1010005791) there about why the accuracy in the Z direction is tricky sometimes due to rounding. 

What do you think? Is your bug the same as the mentioned feature request? Or is it something different?

upatamby, 2023-05-29:
Sorry, I didn't know that I had to reply in this thread, because I received an email from Marimakes, and I just replied that email.
So, here is my reply:

Hi,

thanks for your attention to this issue.
You ask if I expect the first "big overhang layer" to be thinner than it is. Yes I do. I expect that this layer is thinner and also that the layers below this layer will gradually be thinner, anticipating on the requiered layer thickness. The idea of the Adaptive Layers Topography Size is that the distance between the edge of two adjacent layers does not exceed this value, but here it definitely does.

I looked at issue #11160. I'm not sure if this issue is related with my issue. To find out I looked at the model: it has a flat face just above the lowest winding of the worm. See picture:
[image]

So I changed the model a bit to removed that face. That didn't solve anything. As you can see, the layers at the bottom part of the worm are still too thick. The bottom should behave exactly like the top part of the worm.
My conclusion is that my issue is not the same as issue #11160, but it could be related though.

[image]