Ultimaker/Cura #15304: (5.3.0) Slicing fails with segmentation fault in one particular scenario

Vidhuran, 2023-04-20:
### Application Version

5.3.0

### Platform

Mac

### Printer

Ultimaker 3

### Reproduction steps

1. Import the provided model 
2. Set Support Extruder to Extruder 2
3. Start Slicing

### Actual results

The slicing fails and in the CuraEngine debug logs it shows a segmentation fault. The slicing however goes through when selecting Extruder 1 as Support Extruder.

### Expected results

Slicing should not fail when using Extruder 2 as Support Extruder

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[bracelet_large.stl.zip](https://github.com/Ultimaker/Cura/files/11291039/bracelet_large.stl.zip)
[cura-engine.log](https://github.com/Ultimaker/Cura/files/11291041/cura-engine.log)
[cura.log](https://github.com/Ultimaker/Cura/files/11291042/cura.log)

Vandresc, 2023-04-28:
Hi @Vidhuran, unfortunately I cannot reproduce the issue described by you with only the .stl file, could you please also share a project file so I can recreate your configuration? Thank you!

Vidhuran, 2023-04-28:
Hi @Vandresc, thanks for checking this.  I've attached the project file with which you should be able to reproduce the issue. 

[bracelet_large.3mf.zip](https://github.com/Ultimaker/Cura/files/11355124/bracelet_large.3mf.zip)

MariMakes, 2023-05-05:
Hey @Vidhuran,

Thanks for your report 👍
That's a pretty cool model 🤩

I was able to reproduce your crash with the project file. It seems similar to reports that we've already having since one of the more common workaround seems to be moving the model on the buidlplate. 

I was able to slice your model by moving it a bit. Hopefully, that helps you get unstuck. 😇 
[image]

Reports like this can help us us with troubleshooting and fixing issues. 
Thanks 😊 I'll have one of the developers take a look.

Vidhuran, 2023-05-06:
Hi @MariMakes, 

Thanks for checking this. 
The model is from https://www.thingiverse.com/thing:288182. It was designed by Dr. Laura Taalman, who has loads of such cool designs at https://mathgrrl.com/designs/.

Thanks for the tip on moving the model around to get it to slice properly, that worked well and I was able to get the model to slice properly 👍 

There's another interesting finding in relation to this issue. I upgraded Cura to 5.3.1 and was able to reproduce the same behavior with Normal Support Type, but with Tree Support Type the slicing works fine without having to move the model around. Here's the screenshot and a project file with working settings. 

[image]

[bracelet_large_working.3mf.zip](https://github.com/Ultimaker/Cura/files/11412768/bracelet_large_working.3mf.zip)

Tree support is ideal for this model, looks like it'll save me from having to pull out a lot of support material 😄