Ultimaker/Cura #15306: Pause at layer height issue

alexdoni, 2023-04-21:
### Application Version

5.3.1

### Platform

Windows 11

### Printer

Ender 3 V2 / Ender 3 Pro

### Reproduction steps

Added the post processing script 'pause at height', used M0 command, park print head (0,0), standby temp 210.

[image]

### Actual results

After printer was paused, changed the filament, pressed resume in klipper and for some reason klipper was saying 'printer not homed' and couldn't print anymore. See bellow output from console.

[image]

### Expected results

Followed same steps on version 5.2.0, no issue at all, printer stops at the specified layer height, changing the filament, resuming the printer, and it carries on with the print.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura_5.2.0.log](https://github.com/Ultimaker/Cura/files/11291779/cura_5.2.0.log)
[cura_5.3.1.log](https://github.com/Ultimaker/Cura/files/11291780/cura_5.3.1.log)

GregValiant, 2023-04-21:
Thanks for the report.
There were a couple of changes to PauseAtHeight.  One of them was the "Disarm Timeout" that now also has a checkbox "Keep Stepper Motors Engaged".  If the box is is checked then the "Disarm Timeout" defaults to 1 hour.  When it is unchecked the old "Disarm Timeout" box is enabled and you can enter your own time (in Seconds).  The default for the "Disarm Timeout" is "0".  Some firmware interprets "M84 S0" as "Never disable the steppers" while other firmware interprets it as "Disarm Now".  After the steppers disarm they lose their position and that may be why Klipper balked at restarting.
With the "Keep Stepper Motors Engaged" box checked you should see this in the pause code in the gcode file:
M84 S3600 ; Keep steppers engaged for 1hour
M300 S440 P500 ; Beep
M0 ; Do the actual pause
You can open the gcode in a text editor and search for "M0".  Let us know what you find.
There may be a problem with your settings regarding the disarm timeout, or the way Klipper is interpreting either the disarm timeout, or the M0 command.

I don't see a bug here though.  When I use Pause At Height it works as designed.

alexdoni, 2023-04-21:
Yeah, the only difference in gcode between the versions is the M84 S3600 command; when having the "keep motors engaged" box checked. I've unchecked the box and it works fine now.

Apparently in klipper the M84 or M18 command disables all steppers.

closed 2023-04-21 by alexdoni