Ultimaker/Cura #15307: Unexpected failure in slicing.

rusty0101, 2023-04-21:
### Application Version

5.3.0

### Platform

Linux - PopOS 64 bit nvidia 'Jammy"

### Printer

ender3 pro w/ 0.6mm nozzle

### Reproduction steps

Downloadec ball camera mount i created on tinker CAD at https://www.tinkercad.com/things/eXMMhiIHLsB?sharecode=c5Sk4iiueERIvJEYDb1b37oQxRjWb7h5mElob5T0u1c as an stl file. (did not include other parts used in working out dimensions, and adding features.)

imported (dragged and dropped) resulting STL into Cura.

Pressed 'slice'

### Actual results

I get presented with an ''Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker'.

Tested also in 5.0.0 and 5.2.1, both produce similar results. Possible memory exhaustion, though then I would expect it to start using swap, and that has 4 gig available.

### Expected results

I 'expect' Cura to offer to preview the sliced output, or save it, with saving a g-code being the "default' option (if Ihit enter.)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[BallCameraMount.zip](https://github.com/Ultimaker/Cura/files/11291833/BallCameraMount.zip)

rusty0101, 2023-04-21:
Did a reboot. Attempting to load the 3mf project file created as part of this comes up as an invalid file. Loading the same .stl file as indicated comes back with the same error. Loading a file that did work, 

That file did slice, including the project file here to provide settings.
[TensorMicBoomCableClip.zip](https://github.com/Ultimaker/Cura/files/11291882/TensorMicBoomCableClip.zip)

MariMakes, 2023-05-05:
Hey @rusty0101,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 
Wow, that's a super tricky model you are trying to print. I'm not sure you can print it with an FDM printer with a nozzle this big. 

I was able to slice your model with the default settings so that made me thing it's related to one of your settings. 
Indeed when I changed the Minimum Wall Line Width back to the default I was able to slice your model. 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore since it's related to your setting decisions. 
Would you still like to keep this issue open or do you agree that we can close it?

rusty0101, 2023-05-05:
I think we can close it, perhaps with a feature request to have an error
for this situation indicating that the issue is wall thickness. If you
looked at the other model in that tinker cad link, I came up with another
solution that I'm using.

With a better error code, I may have been able to go back and re-model that
part with a larger outer sphere. I suspect it would have worked if the
outer sphere were another 1-2 mm radius larger. As it is, I'm good with the
other holder I created

I've run into the same error with something that I think failed for a
different reason, but I wasn't expecting to build that one out in any case.
I took two 30 mm spheres out of tinkercad, set them to be in the same
position in Cura, then rotated one so that the vertices of it were between
the vertices of the other. It gave me the same error. It may also be an
artifact of using a 0.6 mm nozzle, Or it could have been that in slicing
there were problems with the intersections of the surfaces of the two. I
did not make either hollow, nor attempt to use one to hollow out the other.
I could be wrong on this being an entirely different problem, but it shows
some of the confusion as both attempts resulted in the same error report.

Thank you for looking into the issue, and have a wonderful weekend.

-Rusty

MariMakes, 2023-05-08:
Hey @rusty0101,

Thanks for the update 👍 
I'll close the ticket. 

Good luck with your project.

closed 2023-05-08 by MariMakes