Ultimaker/Cura #15309: bug, unable to slice

skvjlm, 2023-04-21:
### Application Version

5.3.0

### Platform

MacOs Ventura

### Printer

Ender 3 PRO

### Reproduction steps

loaded a moon, tried to slice, got error, unable to slice

### Actual results

no slicing

### Expected results

slicing

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nope

skvjlm, 2023-04-21:
[image]

casperlamboo, 2023-04-24:
Hello @skvjlm, 

Thank you for opening the issue.

Without a project file we are unable to help you. You can share a project file by saving the project (ctrl+s) and creating a zip of the exported 3mf file. The zip can then be uploaded to this issue on github.

The error looks a bit like a known issue where certain orientations of the model result in a slicing error. We are actively working on resolving this issue but there is a workaround. By slightly rotation and/or moving the model Cura can often still slice the model.

Hope this helps!

- @casperlamboo

skvjlm, 2023-04-24:
Well Thanks for the response!It’s just a very low priority project and I assumed it’s because of the really difficult mesh with lots of surfaces.

MariMakes, 2023-05-05:
Hey @skvjlm,

This mesh indeed looks difficult, but not impossible. I've seen people print it before. Where did you get the model? 

If you want us to help you troubleshoot. I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

closed 2023-05-29 by github-actions[bot]