Ultimaker/Cura #15310: Support missing in one out of four identical features of a model

DrRob, 2023-04-21:
### Application Version

5.3.0

### Platform

macOS 11.7.4

### Printer

A heavily modified Creality Ender 5

### Reproduction steps

1. Designed a model in Fusion 360 of a lid for a box with 3mm holes in four corners with a 6mm step near the surface for the head of a hex bolt to recess into
2. Sliced the model in Cura with the 6mm step facing down (because it's the flattest side of the model)

### Actual results

Three of the four holes had supports added.

### Expected results

All four holes should have had supports added, to support the 1.5mm "lip" that's suspended in the air.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log.zip](https://github.com/Ultimaker/Cura/files/11293198/cura.log.zip)
[image]
[lid_bug.3mf.zip](https://github.com/Ultimaker/Cura/files/11293218/lid_bug.3mf.zip)

GregValiant, 2023-04-23:
Thanks for the report and the file.
I think your "X/Y Distance" of 1.2mm is right on the cusp of what tree support will put in those holes.  Regular supports would be a much better choice.  Since the support doesn't have to "reach over" anything then Tree supports aren't indicated for this model.
[image]

There are 73 support settings and by tweaking some of them you can definitely dial in the tree supports for this.  This is with Tree support with the X/Y distance at 0.8.  There is hardly any "trunk" for the tree.
[image]

I don't see a bug here as there are settings available to get this to work with tree support, but regular supports are much better for the conditions you have there.  I'll go ahead and remove the bug label but I'll leave this open so you can respond if you think it is premature.

DrRob, 2023-04-23:
Hi Greg,

Thanks for your informative response, I'll take a look at those options. It's slightly puzzling to me that the supports are generated differently for four features of the model that are all the same, but I won't lose any sleep over it.

Thanks again for looking into it.
Rob.

closed 2023-04-23 by DrRob