Ultimaker/Cura #15311: Cura flatpak 5.3.1 crashes entire system when clicking on preview

iroCheese, 2023-04-21:
### Application Version

5.3.1

### Platform

Fedora 37

### Printer

Ender 3 V2 (Klipper)

### Reproduction steps

1. Add any .stl
2. Slice it (I have automatic slicing enabled)
3. Click preview

### Actual results

Entire system crashes with black screen and colored artifacts similar to broken VRAM. 
My GPU works fine with other software though and I didn't have that issue with previous versions of cura.
[cura.log](https://github.com/Ultimaker/Cura/files/11293558/cura.log)

### Expected results

Preview should be rendered normaly

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Project file not included because the issue occurs no matter which .stl I open.

Might be related to [issue 15263](https://github.com/Ultimaker/Cura/issues/15263).

If it's a flatpak permissions issue here are my settings in flatseal (should all be default):
[image]
[image]
[image]

fieldOfView, 2023-04-22:
Can you try the AppImage instead? The flatpak is not something that Ultimaker maintains; it is packaged by a third party.

iroCheese, 2023-04-23:
I launched the 5.3.1-modern AppImage and it seems to work without the issue. However I would much appreciate installing Cura from some sort of self-updating package manager. I'll contact the maintainers of the flathub repository.
Thanks for your help!

fieldOfView, 2023-04-23:
Thanks for testing the AppImage and reporting back. Since this seems to be a bug with a 3rd-party distribution of Cura, that the developers here are not involved with, I will close this issue.

closed 2023-04-23 by fieldOfView