Ultimaker/Cura #15313: Cura forcing me to send personal information!!

gpgib6, 2023-04-21:
### Application Version

5.3.0

### Platform

Windows 8.1

### Printer

Not telling you

### Reproduction steps

Install Cura.

### Actual results

There was a dialog that said personal information such as which operating system I use, which printer I have and who knows what else, probably my IP address, my approximate location, I'm sure the list goes on. The point is, it should be my decision whether to share this information. Should I not have the right not to tell you about my setup?
Unfortunately I had to uninstall the program before even trying it because of this reason. I still want to try it but now I have to set up a pfSense machine just for that PC to block the internet and still allow my remote desktop. Only then I can reinstall the program. 

### Expected results

You can fix this bug and add a box that says "Yes, please share my information", or at the very least (!) add a box that says "don't share my information".

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional information is available.

fieldOfView, 2023-04-22:
I think you mean this dialog:
[image]
If you press the "More information" link, you will be given the option not to send the information. That dialog also shows an example of the type of information that is sent. I agree the option is a bit hidden, but it is there. 

If you missed that option, you can still turn of the sending of information in the General preferences in Cura; scroll all the way down to the Privacy section and untick the "Send (anonymous) print information".

Please let us know if this solves your issue.

gpgib6, 2023-04-24:
Thank you I found it. I'm not a fan of hiding this setting but at least it's there.

closed 2023-04-24 by gpgib6