Ultimaker/Cura #15314: fehler beim slicen

Danuell9090, 2023-04-21:
### Application Version

5.3.0

### Platform

windows 10

### Printer

anycubic mega s

### Reproduction steps

ich wollte eine datei vergrössern und slicen .
das slicen hat nicht funkioniert wenn ich die datei grösser gemacht habe 

### Actual results

hallo
sobald ich eine datei vergrössere und slicen möchte , kommt nach den slicen immer " UNERWARTETER FEHLER"

bitte helfen sie mir 
mein drucker : Anycubic Mega S

### Expected results

das er fertig slicen tut 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]

Danuell9090, 2023-04-24:
[AI3MSPRO_Hex3D_Skull2_Bowl.zip](https://github.com/Ultimaker/Cura/files/11312885/AI3MSPRO_Hex3D_Skull2_Bowl.zip)

Danuell9090, 2023-04-24:
> Hi @Danuell9090<https://github.com/Danuell9090>
>
> thank you for opening the issue. My German is not so good so excuse me for replying in English.
>
> Without a project file we are unable to help you. You can share a project file by saving the project (ctrl+s) and creating a zip of the exported 3mf file. The zip can then be uploaded to this issue on github.
>
> The error looks a bit like a known issue where certain orientations of the model result in a slicing error. We are actively working on resolving this issue but there is a workaround. By slightly rotation and/or moving the model Cura can often still slice the model.
>
> Hope this helps!

Hallo
Ich habe ihnen eine ZIP DATEI geschickt

MariMakes, 2023-05-08:
Hey @Danuell9090,

Thanks for the Project File 👍 It seems to be different from the screenshot you shared. But I'm also getting a crash. 
I took a look at it in Cura 5.3.1 and it seems to be related to your layer height. 
When I reset that value from 0.4 to 0.2 I can successfully slice your model: 

[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore since you changed the setting value into a warning zone. 
Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-05-29 by github-actions[bot]