Ultimaker/Cura #15316: Unexpected error when multiplied model and proceed for slicing.

Ananyapranav, 2023-04-21:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Prusa 3D  i3

### Reproduction steps

1. I imported a stl file.
2. I scaled the object uniformly to get height to 50 mm.
3. I selected the object and from right-click menu selected multiply option and set the value to 14(to get total 15 copies of the selected object).
4. I then re-arranged the object using Arrange All Models and tried slicing again (but got the same error).
5. I started removing the objects one by one and tried slicing each time (but got same error every time).

### Actual results

As I kept deleting objects one by one, finally when only 3 objects were left, it sliced them successfully.

### Expected results

I expected to get total 15 objects sliced so that I could print 15 objects in 1 print cycle.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CCR20_AstronautsitzendProject.zip](https://github.com/Ultimaker/Cura/files/11298248/CCR20_AstronautsitzendProject.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/11298266/cura.log)

casperlamboo, 2023-04-24:
Hello @Ananyapranav,

Thank you for opening the issue.

Without a project file we are unable to help you. You can share a project file by saving the project (ctrl+s) and creating a zip of the exported 3mf file. The zip can then be uploaded to this issue on github.

The error looks a bit like a known issue where certain orientations of the model result in a slicing error. We are actively working on resolving this issue but there is a workaround. By slightly rotation and/or moving the model Cura can often still slice the model.

Hope this helps!

Ananyapranav, 2023-04-25:
Yeah sure I'll update the project file.

Ananyapranav, 2023-04-26:
please find attached
[CCR20_AstronautsitzendProject.zip](https://github.com/Ultimaker/Cura/files/11332650/CCR20_AstronautsitzendProject.zip)
 the project file

MariMakes, 2023-05-05:
Hey @Ananyapranav,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for the file👍 That's a pretty cool model 🤩

I think I know what is going wrong. You have Union Overlapping Volumes unchecked. 
If you enable that setting you can slice your buildplate with astronauts. 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

MariMakes, 2023-05-08:
Awesome, Thanks for the update. Happy printing 🚀

Ananyapranav, 2023-05-08:
Thanks for your support!