Ultimaker/Cura #15317: Incredible error when print a object (and only thtat)

MarcelloMazza, 2023-04-21:
### Application Version

5.3.0

### Platform

Window 10

### Printer

FLSUN V400

### Reproduction steps

I launch a print of the same large but not very complex object, but at the second layer there is a trouble.

### Actual results

At the second layer, when the head reaches about halfway, an overextrusion occurs and the head descends until it removes the material from the first layer and I have to stop printing. All this does not depend on the orientation of the piece.
Layer heght first and second 0.2 mm.
[image]

### Expected results

Stop printing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

In the gcode file there are not problem and the trouble is only for that piece

nallath, 2023-04-25:
From the information that you have provided here, this doesn't seem to be an issue with Cura. It looks much more like the bed isn't leveled correctly. Could you try leveling the bed and try again?

MariMakes, 2023-05-05:
Hey @MarcelloMazza,

I'm tending to agree with @nallath, this seems to be a leveling issue. 
I'm seeing SnailTracks, which often means that the nozzle is too close. 

[image]

To me, it doesn't seem like a bug in Ultimaker Cura but one with your printer. 
You can reach out on Reddit or facebook to other FLSUN users or even start a discussion on our [UltiMaker community forum which you can find here.](https://community.ultimaker.com/). 

I'll be closing this issue. I hope you understand. 
Good luck!

closed 2023-05-05 by MariMakes