Ultimaker/Cura #15319: errno15!!!!

lamaquettedetrain, 2023-04-21:
### Application Version

5.2.2

### Platform

windows10

### Printer

creality ender 3 pro

### Reproduction steps

j'ai voulue enregistrer une impression dans cura mais un code d'erreur est apparue portant le nom de errno 15

### Actual results

aucun changement

### Expected results

resoudre le probleme

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

jhjjjn,hjhgn,jhg,nj,

casperlamboo, 2023-04-24:
Hi @lamaquettedetrain, 

Thank you for opening the issue. My French is not that good so pardon me for replying in English.

Without the information currently provided it's a bit hard to help and resolve the issue for further users. If possible could you please share your
- log files (you can follow this link to learn where to find the log files https://support.makerbot.com/s/article/1667337559413). It would help a lot if you could trigger the bug, and once cura crashes immediately share a copy of the logs. This saves us a lot of time investigating where in the log the error occurred.
- Additionally (although less important) could you also share the project file?

MariMakes, 2023-05-05:
Hey @lamaquettedetrain,

Welcome to the UltiMaker Cura Github 🚀 

The new Cura 5.3 Stable contains the fix for this for a number of slicing crashes. 
Could you test if that's the case for your model as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

Have you tried reverting to the default settings?
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-05-29 by github-actions[bot]