Ultimaker/Cura #15322: Surface works perfect with 4.13.1 but 5.3 does not

Tymat324, 2023-04-22:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Custom (flashforge guider 2s)

### Reproduction steps

Atempt to use slicing tools in viewport rotate zoom scale etc

### Actual results

Stylus or pen does not work properly on my micosoft surface mainly in viewport rotate zoom move etc worked fine in 4.13.1 ansd works in 5.3 until i select object or select to rotate view and then cant release from that action

### Expected results

To work like it did in previous releases

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

casperlamboo, 2023-04-24:
Hi @Tymat324 ,

Thank you for opening the issue. It looks like the issue you opened is a duplicate of https://github.com/Ultimaker/Cura/issues/12554. Could you confirm you are indeed using a wacom tabled?

Tymat324, 2023-04-26:
No im using a Microsoft surface 7pro

MariMakes, 2023-05-08:
Hey All,

The ticket that @casperlamboo mentioned also includes other touchscreen devices, so even if it's on a Microsoft Surface I would consider it to be a duplicate of https://github.com/Ultimaker/Cura/issues/12554

We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12554

Thanks for your report 👍

closed 2023-05-08 by MariMakes