Ultimaker/Cura #15323: Ultimaker_Cura-4.13.1.AppImage doesn't work with Ubuntu 18.04

PSLLSP, 2023-04-22:
### Application Version

4.13.1.

### Platform

Ubuntu 18.04.6 LTS

### Printer

NA

### Reproduction steps

Ultimaker_Cura-4.13.1.AppImage doesn't work with Ubuntu 18.04.1:

```
$ ./UltiMaker-Cura-4.13.1.AppImage 
cura: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by cura)
```

This is bad change in minor release. `Ultimaker_Cura-4.13.0.AppImage` was OK. `Ultimaker_Cura-4.13.1.AppImage` was the last release that could be compatible with Ubuntu 18.04 but you broke that compatibility :-(

### Actual results

NA

### Expected results

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

NA

nallath, 2023-04-25:
You're right that it's bad to do that, but due to significant issues with our build-system, it was the only solution that we had. 

We strive to support as many systems as possible, but this is pretty hard to do. Since we don't have unlimited resources, we sometimes have to make tough choices regarding how much we want to spend on supporting older systems. In this case, that meant accepting that certain operating systems didn't work any more.

closed 2023-04-25 by nallath

eGit, 2023-09-17:
Can anybody please make a backwards compatible appimage ONCE for 5.4 or newer, so people stuck on older versions of GLIBC can benefit from a newer version of Cura?

Many people are still on GLIBC_2.26 (NOT Ubuntu 18.04, but it has the same GLIBC version). My Linux is still not at its End of Life for a long time, so I can't just update it JUST because Cura is so very stringent.

Please make at least ONE appimage of 5.x