Ultimaker/Cura #15326: cura 5.3 v400 support error

sadamop, 2023-04-22:
### Application Version

5.3

### Platform

windows 10 home 

### Printer

flsun v400

### Reproduction steps

when generating a support does not start to the first layer, but part to the second layer consequently does not go
[image]
[image]

### Actual results

nothing because the supports so you can not use

### Expected results

will solve the problem soon

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothing to add

GregValiant, 2023-04-22:
Thanks for the report.
We really need a project file to investigate this.  Load the model and use the "File | Save Project" command.  Zip the file and post the zip folder here.

sadamop, 2023-04-23:
Hi the problem is not the file because in cura 5.2.2 this generate correctly the support the problem is all cura 5.3 i have much friends 
with the same problem on all files and unfortunately we have to use version 5.2.2 but there is no flsun v400 profile and we have problems
Inviato da Libero Mail

casperlamboo, 2023-04-25:
Hi @sadamop,

The problem is indeed most likely in the program. However for us to properly debug the issue we need the file in order to properly see what is going on, and to see if the fix will actually resolve this problem.

To create the project file click safe from the file menu (or press `ctrl+s`). The create a zip from the exported 3mf file and share this zip on this page on github. Once this is done we can properly resolve the issue.

Thank you in advance.

sadamop, 2023-04-26:
[problem support.zip](https://github.com/Ultimaker/Cura/files/11330887/problem.support.zip)

GregValiant, 2023-04-26:
Thanks for the file.
@casperlamboo I can duplicate this when the Bed Adhesion is set to "None" and the "Enable Support Brim" is checked.
@sadamop A temporary fix is to set the Bed Adhesion to Skirt or Brim and/or to disable the Support Brim.

sadamop, 2023-04-26:
ok tnx 
ok now I saw, if you do not use any adhesion to the plate the support does not go to the first layer.
 but if you set brim or skirt to the first true layer generated with skirt or brim and the true support above, but I hope you solve this problem

MariMakes, 2023-05-08:
Hey All,

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14983
We have a ticket on our backlog with the intent to improve this behavior.

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14983
Thanks for your report 👍

closed 2023-05-08 by MariMakes