Ultimaker/Cura #15328: Acceleration Tower Problem

fabiobarros79, 2023-04-22:
### Application Version

5.2.2

### Platform

Windows 10 Pro

### Printer

Ender 3 S1 Pro

### Reproduction steps

Open Cura
Extensions
Part for Calibration
Add an Acceleration Tower
Select the piece by clicking on it.
Press R to rotate
Press blue arrow to change orientation (to match X of piece with X ok printer)

At this point, if I click on slice

This error appears

[image]

If I don't change the orientation, there's no problem.

### Actual results

Slicing error

### Expected results

Normal slicing

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11302264/cura.log)
[Error project.zip](https://github.com/Ultimaker/Cura/files/11302265/Error.project.zip)

GregValiant, 2023-04-22:
Thanks for the report and the files.
I can't get the model to "not slice" so I'm unable to duplicate the problem.
The plugin was written by @5axes so you might want to give him a heads-up on his GitHub page.
There have been several other reports of Cura having trouble slicing models and this may be a duplicate of that bug rather than something wrong with the plugin.  I'll mark this as a slicing error and I'm sure both the Cura team and 5Axes will take a look.

5axes, 2023-04-23:
https://github.com/5axes/Calibration-Shapes/issues/106

fabiobarros79, 2023-04-23:
> 5axes/Calibration-Shapes#106

Hi there,

I Just updated to Cura 5.3 and the problem was solved.

Thank you very much.

GregValiant, 2023-04-23:
@fabiobarros79 I'll go ahead and close this as the new version seems to be the fix.  Let me know if you think that is premature.

closed 2023-04-23 by GregValiant