Ultimaker/Cura #15330: Slicing failed with an unexpected error.

Terrybi, 2023-04-23:
### Application Version

5.3.0

### Platform

Win 11

### Printer

Creality Ender 3 s1 Pro

### Reproduction steps

1. Loaded STL
2. Started slicing

### Actual results

Slicing failed with an unexpected error.

### Expected results

[cura.log](https://github.com/Ultimaker/Cura/files/11303797/cura.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Tried slicing file with pursa slicer and it worked fine.

GregValiant, 2023-04-24:
Thanks for the report.
"Backend exited abnormally with return code 3221225477!"
My understanding is that the code is a memory error from the Cura Engine.
It would be helpful if you would zip and attach a Project File.  This might involve a model error.  (PrusaSlicer has an auto-run repair routine.)

Terrybi, 2023-04-24:
Thanks for the reply. Attached is a zip of the file that failed. This is the first time I've had an error like this.
[duck2sculpt.zip](https://github.com/Ultimaker/Cura/files/11313864/duck2sculpt.zip)

Terrybi, 2023-04-24:
Just for interest, I loaded up and sliced six other stl's I have and they all sliced no problem at all.

GregValiant, 2023-04-24:
I didn't have any trouble slicing the duckbill model either.  I tried a bunch of different things and couldn't get it to "not slice.
I'll leave this as a bug report and the Cura team will take a look.

Terrybi, 2023-04-25:
OK Thanks for looking. I noticed there are a few others reporting the same error but with different processors. My machine is an MSI "gaming" computer with Intel(R) Core(TM) i5-10400F CPU @ 2.90GHz   2.90 GHz and 16 Gb RAM. The GPU is an Nvidea GeForce GTX 1660 Super. Don't know if that info helps, but here it is.

closed 2025-11-05 by Terrybi

GregValiant, 2025-11-05:
One of the things I've noticed is that computer manufacturers are tending to "over-clock" their machines in order to tell everyone "Heh, we're the fastest".   It seems more prevalent on high-end machines.  The over-clocking seems to cause "overheating" on some systems and when the processor cuts back to protect itself, Cura can be dumped by the OS.

I'm not saying that's your problem, I'm just saying there can be other things at play.  That "duck" is fairly simple (107,000 facets) and as far as I can tell does not have any errors.  It really shouldn't tax the computer BUT it did take 30 seconds to slice on my decidedly "bottom end" laptop.  Maybe it's tougher than it looks.