Ultimaker/Cura #15332: Slicing Failed

FlashMoonUK, 2023-04-23:
### Application Version

5.3

### Platform

Mac OS

### Printer

Dremel 3D45

### Reproduction steps

Had 3 bodies on the plate and tried to slice - it would slice but not open the save window
closed Cura
opened Cura and imported one file
Sliced on exactly the same setting and it failed

### Actual results

Failed - Bug report

### Expected results

Should have sliced
[cura.log](https://github.com/Ultimaker/Cura/files/11304208/cura.log)
[Wahoo Adaptor 3 v10.stl.zip](https://github.com/Ultimaker/Cura/files/11304215/Wahoo.Adaptor.3.v10.stl.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I am having a few issues now with Cura on my MAC - I believe this will slice with windows 10

GregValiant, 2023-04-24:
Thanks for the report and the file.
I'm on Windows 11 and didn't have any trouble slicing your project.  The model doesn't have any errors.
Someone from the Cura Team will take a look.

casperlamboo, 2023-04-24:
Hi @FlashMoonUK, 

Thank you for opening the bug report. I can indeed reproduce the error. The error happens due to a change in the "minimum speed" setting. Since recently we imposed a [minimum value on this setting of 1 mm/s](https://github.com/Ultimaker/Cura/commit/7e194d34b6bf9b55b773426dc2771643c4b115d4). This change does make sense and as far as we could see at the time this did _not_ affect any of the printers/profiles within cura.

However we did not think about external plugins that might also add printers. The Dremel printers have a minimum speed which is set to default to 0 and is added to cura via a plugin. As such have no control over the content of the plugin and cannot change this default value.

I'm going to bring this up with the team and see if we can maybe change the "error" to a "warning" instead so that the model at least slices.

In the mean time I do have a workaround; change the minimum speed setting to "1". This is equal to the minimum value for the minimum speed and as such it slices as expected.

FlashMoonUK, 2023-04-25:
Hi Casper,

Thank you for investigating this anomaly. I can confirm that the minimum speed on my Mac version of Cura 5.3.0 was set to 0.0 mm/s and that changing it to 1.0 enable me to slice the model.  I have previousley been able to slice the same model on a Windows PC running windows 10 and have then printed it, it does NOT print correctly, the top surface and features are lost.  Changing the value on the Windows PC to 1.0mm/s increases the printing time from 12 minutes to 13 minutes.  I will try to print this later today and let you know if I am successful.

Would you mind if I share this information with the Dremel Facebook community please?  Thanks 

Thanks again for you help

Kind regards

Chris