Ultimaker/Cura #15333: Installation failure

JKBert, 2023-04-23:
### Application Version

5.4.0-Alpha

### Platform

Linux Debian

### Printer

Anycubic Viper

### Reproduction steps

Hello,
I'm trying to build Cura from sources. Thus, I have followed indications I have found on github, but installation fails with :
```
ERROR: Generator VirtualPythonEnv(file:None) failed
Error 127 while executing /home/bertrand/.conan/data/cpython/3.10.4/_/_/package/9054a82b60899c3ed90865d5b990fd99ef2dc167/bin/python3.10 -m venv /home/bertrand/git/Cura-build/cura_inst

```

I have installed all required packages and conan 1.56.0 with pip. After this installation, I have lauched
```
conan create . --build=missing --update
git clone https://github.com/Ultimaker/Cura.git
cd Cura
conan create . cura/5.4.0-alpha@me/testing --build=missing --update

```
No error. Now, I'm trying to install software.
```
cd ../
mkdir Cura-build
cd Cura-build/
conan install cura/5.4.0-alpha@me/testing --build=missing --update -g VirtualPythonEnv -o cura:devtools=True -if cura_inst --json cura_inst/conan_install_info.json
```
Last command fails with:
```
----Running------
> /home/bertrand/.conan/data/cpython/3.10.4/_/_/package/9054a82b60899c3ed90865d5b990fd99ef2dc167/bin/python3.10 -m venv /home/bertrand/git/Cura-build/cura_inst
-----------------
/home/bertrand/.conan/data/cpython/3.10.4/_/_/package/9054a82b60899c3ed90865d5b990fd99ef2dc167/bin/python3.10: error while loading shared libraries: libpython3.10.so.1.0: cannot open shared object file: No such file or directory
ERROR: Generator VirtualPythonEnv(file:None) failed
Error 127 while executing /home/bertrand/.conan/data/cpython/3.10.4/_/_/package/9054a82b60899c3ed90865d5b990fd99ef2dc167/bin/python3.10 -m venv /home/bertrand/git/Cura-build/cura_inst

JSON file created at '/home/bertrand/git/Cura-build/cura_inst/conan_install_info.json'
ERROR: Error 127 while executing /home/bertrand/.conan/data/cpython/3.10.4/_/_/package/9054a82b60899c3ed90865d5b990fd99ef2dc167/bin/python3.10 -m venv /home/bertrand/git/Cura-build/cura_inst
```
I don't understand why process tries to use python 3.10. I only have 3.11 on this workstation. How can I force python 3.11 in build process ?

Regards,

JB

### Actual results

Install uses python 3.10.

### Expected results

I only have Python 3.11 and I want to configure Cura to use this version.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

MariMakes, 2023-05-08:
Hey @JKBert,

Welcome to the UltiMaker Cura Github 🚀 

😖 Sorry to hear you are stuck.

I'm only a fledgling member of the Build From Source Club 👑 
Before we get started, I have a few questions to narrow down the troubleshooting:

1. Can you make sure you have specifically Conan version 1.56.0?  
`pip install conan == 1.56.0 --force `
You can check the [Conan Documentation]( https://docs.conan.io/en/latest/installation.html ) for the latest version and installation instructions

2. Did you try clearing the Conan Cache? 
`conan remove "*" -f `

3. Can you make sure you have the latest config with this command?
`conan config install https://github.com/Ultimaker/conan-config.git`

4. Can you make sure you have a default profile?
`conan profile new default --detect --force`

5. With these steps you should have a clean slate and you can follow the steps on our wiki again. 
[You can find it here. ](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source) 

If you are stuck in any way, please reach out. We are happy to help out. 
The best way to do that is to provide us with your complete run-out log. 
Not just the part that fails, but all the lines generated after you've given your command. 
We need it see your profiles and packages. 

I'm super curious. What do you want to achieve by creating a Build?
Are you intending to share it with others or are you using it for research purposes?
I would love to know more about your project, maybe we can help.a

valba, 2023-05-10:
I had the same problem than JB.
I solved it by exporting the correct LD_LIBRARY_PATH.

1. Execute asdf
2. 
3. 

From the example above, Bertrand would solve it executing this command just after the error:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/bertrand/.conan/data/cpython/3.10.4/_/_/package/9054a82b60899c3ed90865d5b990fd99ef2dc167/lib
And continue the building again with
 with the following command

valba, 2023-05-17:
I had the same issue reported by JKBert.
I solved it by exporting the correct LD_LIBRARY_PATH just after the error arose.
Although I have python 3.11 installed in Fedora 38, I used _anaconda_ to create a safe python environment for version 3.10.4.

So, I followed MariMakes' directions but I had to add LD_LIBRARY_PATH in order to be able to recompile Cura over and over again.

I detail the steps followed here under:

1. Install development packages:
`dnf install ninja-build libxcb-devel libfontenc-devel libXaw-devel libXcomposite-devel libXcursor-devel libXdmcp-devel libXtst-devel libXinerama-devel libxkbfile-devel libXrandr-devel libXres-devel libXScrnSaver-devel libXvMC-devel xorg-x11-xtrans-devel xcb-util-wm-devel xcb-util-image-devel xcb-util-keysyms-devel xcb-util-renderutil-devel libXdamage-devel libXxf86vm-devel libXv-devel xcb-util-devel libuuid-devel xkeyboard-config-devel ip-build python3 venv`

2. Use _anaconda_ to create a safe python environment for version 3.10.4:
`conda create -n cura python=3.10.4`

3. Activate the conda environment: 
`conda activate cura`

4. Install the following python packages:
`pip install virtualenv sip decorator ipython pexpect async-generator`

5. Follow directions from https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source
`pip install conan == 1.56.0 --force
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force`

6. Go to the place from where you want to install Cura:
`cd $HOME/local_apps`

7. Clone Cura:
`git clone https://github.com/Ultimaker/Cura.git 
cd Cura`

8. Flush the conan cache: 
`conan remove "*" -f`

9. Build Cura:
`# optional for a specific version: conan install . cura/<version>@<user>/<channel> --build=missing --update`
`conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv`

10. In case you find the same error reported by JKBert, you must delete the _venv_ directory in case it were created previously:  
`rm -r ./venv`

11. Then, export the LD_LIBRARY_PATH variable. From JKBert example it should be something like this: 
`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/bertrand/.conan/data/cpython/3.10.4/_/_/package/9054a82b60899c3ed90865d5b990fd99ef2dc167/lib`
In my case, it was the following:
`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/valentin/.conan/data/cpython/3.10.4/_/_/package/175731c75565d7692aafd6c09d0bab517aa02934/lib`

12. Verify that it was exported correctly:
`echo $LD_LIBRARY_PATH`

13. And finally, repeat step 5 to continue building Cura:
`# optional for a specific version: conan install . cura/<version>@<user>/<channel> --build=missing --update`
`conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv`

Every time I have needed to recompile Cura again, I had to start the same sequence of steps, but starting from step 5.

Btw, I had to write the following script to launch Cura properly:
`cat $HOME/local_apps/launch_cura.sh`

`#!/bin/bash`
`cd /home/valentin/local_apps/cura/Cura`
`source /home/valentin/local_apps/cura/Cura/venv/bin/activate`
`python /home/valentin/local_apps/cura/Cura/cura_app.py`
`source /home/valentin/local_apps/cura/Cura/venv/bin/deactivate_activate`

I hope it helps!

JKBert, 2023-05-20:
Thanks a lot, it now runs as expected.
Regards,
JB