Ultimaker/Cura #15336: atat helmet will not complete slice

juztinh, 2023-04-23:
### Application Version

5.3.0.beta 2

### Platform

windows 10 pro

### Printer

lk5pro

### Reproduction steps

imported file, positioned so it would turn yellow, hit slice and error

### Actual results

failed to slice

### Expected results

would love this to actually slice, slices fine in prusa

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[LLK5PRO_Domeerror.zip](https://github.com/Ultimaker/Cura/files/11304882/LLK5PRO_Domeerror.zip)

GregValiant, 2023-04-24:
Thanks for the report and the file.
I couldn't get your model to slice either.  When I checked it with the Cura Mesh Tools it showed as "not watertight".
The model has errors and can confuse Cura.  Here is the report from an on-line STL repair site:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 0 Non-manifold edges (?)
--> 0 Inverted faces (?)
**--> 8 Degenerate faces (?)**
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 17074ms ------
**-> Vertex count changed from 60654 to 61262 (+608)
-> Triangle count changed from 121312 to 122716 (+1404)**

I don't know what a "degenerate" face is but it doesn't sound good.
I repaired the model and it sliced.  With your settings it's a 126hour print and a full 1kg roll of filament.
I'll leave the bug label on this for a bit and the Cura Team will review this but the model errors make it really hard to tell if this is actually a bug, or a normal response to a faulty model.
[image]

juztinh, 2023-04-24:
Sounds good, I did not mess with other settings as far as infill,
retraction or others to lower the time, prusa on my settings at a .15 line
15% infill etc has it at a 2 1/2 day run I think.  I will have to take a
pic.  Appreciate  you looking into this. Will be interested in how it
goes,  I decided to try prusa after it failed in cura.

juztinh, 2023-04-24:
> PrusaSlicer and IdeaMaker have repair routines that auto-run.
> I'll go ahead and close this then. Model errors always fuzz the line
> between "bug" and the model error so this particular report wouldn't be a
> lot of help in chasing a bug down. I hope you understand.

Yes no issues.  I will just use prusa more, I love the look and feel of
cura as to me it has a more user friendly interface, would love to see cura
have an repair option that could fix minor issues on models. It's sad it
just says fail on my side with no explanation, I should have thrown it in
mesh or something to check for issues. It would be nice to have cura take
that step and do repairs or at least tell me why it's not completing the
slice. If it knows to not complete it would be great to relay to the user
as to why it did not complete, its done it to me on another large helmet
from different person. But thank you for your help. Is cura going to ever
offer repair model or anything like that?