Ultimaker/Cura #15338: Preview bug when moving up on the z-axis

Ak1K0rt, 2023-04-24:
### Application Version

5.3.0

### Platform

Windows 10 and Windows 11

### Printer

Anykubic Kobra

### Reproduction steps

1. Open STL-file in Cura
2. Check "Generate Support" and "Touching Buildplate" settings.
3. Uncheck "Drop Down Model" from the Move ("T") settings
4. Move Z-axis 10mm upwards
5. Press Slice
6. Press Preview

### Actual results

Printable model remains attached to the buildplate.

[image]

### Expected results

Printable model should move upwards 10mm and there should be support between buildplate and model. This screenshot is taken from Cura 5.2.1.
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Cura 5.3.0 logfile

[cura.log](https://github.com/Ultimaker/Cura/files/11307581/cura.log)

GregValiant, 2023-04-24:
Thanks for the report.
I think you have stumbled on to "one of those things".
When "Generate Support" is checked - it hides the "Mesh Fixes" setting "Remove Empty First Layers" BUT "Remove Empty First Layers" is being used (even when hidden) and if it happens to be enabled then yes, the model is printed on the build plate instead of on top of a bed of support.
So the workaround is to:

- Turn off "Generate Support"
- Turn off "Remove Empty First Layers" in Mesh Fixes
- Re-enable "Generate Support"

I feel that a setting that is being used shouldn't be hidden and there is confusion between "Remove Empty..." and the General Setting "Automatically drop models to the build plate".  The Cura Team is looking into this.

This is a duplicate of #14989 and #15179 and I'll mark it as such and I'll go ahead and close this ticket given that the team has indicated it will be addressed.  I hope you understand.

closed 2023-04-24 by GregValiant

Ak1K0rt, 2023-04-24:
Thanks for the tip!