Ultimaker/Cura #15342: (5.3.0) Produces GCode containing NaN values

TripleWhy, 2023-04-24:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Anycubic Vyper

### Reproduction steps

1. Open the attached project file: [AV_35.zip](https://github.com/Ultimaker/Cura/files/11311387/AV_35.zip)
2. Click Slice
3. Save the file.

### Actual results

1. The gcode file contains NAN values.
2. The print time is much too short. (Probably due to 1.)
3. The physical printer will print the first few layers, then continue without extruding.
4. [ZupFe](https://zupfe.velor.ca/) produces the same result as the physical printer.
5. The Cura Preview does produce the expected slicing result, although it should clearly not when there is no extrusion.
6. There is no warning or error message of any kind.

### Expected results

1. A printable GCode file or
2. An error message in the UI that tells the user that something is wrong and how to fix it.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11311451/cura.log)

GregValiant, 2023-04-24:
Thanks for the report.
Are you printing from an SD card and is the problem with a file on the card or on your hard drive?
I can't duplicate the problem with your project file.  I read the Gcode into AutoCAD it it is exactly what I would expect.
Looking through the gcode on the Zupfe site I'm not sure what to make of it.  Does that site edit the gcode?  There is no opening paragraph or StartUp or Ending Gcode or Layer Change code.
The problem you have described is what I would expect from corrupt memory segments on an SD card.  Reformatting usually fixes the problem.
Cura doesn't actually write anything to disc.  Like most apps it makes a call to the operating system which does the actual write to disc.
A comparison between a gcode file on your hard disc, and the same file copied to an SD card may show a size difference when in fact they should be identical.

TripleWhy, 2023-04-24:
First, I thought it was the printer, it wasn't.
Then I thought it was the file or SD card, it wasn't.
Cura actually only generates lines like this after a few layers:
```
G1 F-NAN(IND) X148.015 Y139.74 E-NAN(IND)
```
Here is the file I get when I slice it: [AV_35.gcode.zip](https://github.com/Ultimaker/Cura/files/11314088/AV_35.gcode.zip)

This it what it looks like in ZupFe:
[image]
And here it is with travel moves:
[image]
I chose this simulator, because this is what it looks like on my printer.

Other simulators produce this, for example:
[image]

Cura thinks it is fine when I slice it and look at the preview:
[image]

However if I open just the gcode file in Cura, it will match what I see in ZupFe:
[image]

GregValiant, 2023-04-24:
Alright, I see.
You have the plugin "ArcWelder" running (but that model is all straight lines and so ArcWelder isn't needed and actually doesn't do anything).  Do you have any other plugins or post-processors enabled like the Klipper plugin or Octoprint?
Those gcode lines that are being inserted look like they might be for another printer.  They certainly aren't needed for your AnyCubic and although I look at a lot of gcode files from a lot of different slicers, I'm not familiar with those lines at all.
Lines like:
;FLOW_RATE_COMPENSATED_OFFSET = -nan(ind)
G1 F-NAN(IND) X148.484 Y139.875 E-NAN(IND)
those are not Cura lines.  That G1 line looks like placeholder variables for the FeedRate and the Extrusion Distance that something else is going to put numbers into so the printer would understand it.
You've got something going on there and if those lines are being passed on to the printer without any numbers being added then the printer will skip them as an "Unknown Command" and that's why it isn't printing.  I don't think this is a Cura problem though.

TripleWhy, 2023-04-25:
I noticed that ArcWelder was enabled, too. The result is the same without it though. The last picture is actually without ArcWelder, because Cura can't render arcs for some reason.

I don't think I have any other post processing plugins enabled (or even installed). Is there a way I can verify this or disable post-processing to check?

GregValiant, 2023-04-25:
There are essentially two ways for a user to customize Cura.
One is with "Plugins" that are available in the MarketPlace (the button is in the top right of the Cura workspace).  You would need to go there and scroll down through them and see which ones you have "Installed".  You could also check your Configuration Folder ("Help | Show configuration Folder") and see if the "Plugins" folder contains anything.  Disabling or un-installing all plugins should get you back to an "As Installed" state where Cura just does it's thing.
Some plugins are just too handy to go without.  "Mesh Tools", and "Printer Settings" are two that I always have installed.  All the plugins become inter-woven with Cura and any commands they add are within the Cura interface.

The second way to customize is with Post-Processors that alter the G-Code after Cura has finished slicing, but before it is written to a file.  Those are available under "Extensions | Post-Processing | Modify Gcode".  When any Post-Processors are active there is a button on the left end of the slice button and it will have a red circle with a number in it.  The number indicates how many post processors are active.  Click on the button and then click on the "X" for each post processor that might be active.
Unlike plugins, post-processors are not inter-woven into Cura.  They are meant for users to be able to make some changes to the gcode for things like Pauses and Filament Change that could be hand coded, but it's way easier when it is automated.

You have something running that is affecting the Gcode that is produced.  All the [Cura Keywords are here](http://files.fieldofview.com/cura/Replacement_Patterns.html) and those -NAN(IND) lines aren't in there.  That's why I suspect that you have some other process going on.

TripleWhy, 2023-04-25:
I don't have any post processors installed.

Also note that `-NAN(IND)` is not a keyword, but an [actual value](https://en.wikipedia.org/wiki/NaN) that a [floating point](https://en.wikipedia.org/wiki/Floating-point_arithmetic) number can have. NaNs can have a sign, thus the minus in front of it. There are different kinds of NaN, and the `ind` part might indicate one, but it's not obvious to me what it is supposed to mean.

NaN usually indicates that some computation went wrong. So my hypothesis is: Cura does its thing, but somewhere there is a division by 0 for example. There is no check for that after the computation, and it just writes it to the gcode file like it would if the result was a regular number.

This isn't completely true: Cura seems to notice it, but instead of producing an error, it puts an additional comment before each line containing NaN. These lines only appear once we get to the NaN layers:
```
;FLOW_RATE_COMPENSATED_OFFSET = -nan(ind)
```

TripleWhy, 2023-04-25:
Alright, I did the following:

1. Deleted AppData\Roaming\cura\5.3
2. Opened Cura
3. Selected the Anycubic Vyper printer
4. Opened the .3mf file I attached to the report
5. Clicked slice
6. Saved the file.

Here it is: [AV_35_clean.zip](https://github.com/Ultimaker/Cura/files/11324169/AV_35_clean.zip)
As you can see, the problem still occurs. There are no manually installed plugins, this just the clean default cura configuration.

fieldOfView, 2023-04-25:
I retract both my comments. I can reproduce the nan(ind) in the gcode with your project file and without any plugins that could cause this installed.

The `;FLOW_RATE_COMPENSATED_OFFSET = -nan(ind)` line is created here:
https://github.com/Ultimaker/CuraEngine/blob/90666835f5b2330ab5058574e35683d86c82b984/src/gcodeExport.cpp#L970
After that the `current_e_offset` is no longer valid, so all subsequent E-values are invalid (not numbers).

This does look like a bug in CuraEngine.

fieldOfView, 2023-04-25:
The problem is that you set the `Minimum Speed` to 0. I think Cura should give you an error when you do that, because covering any distance with a speed of 0 is going to be problematic.

Set the `Minimum Speed` to a value that is not 0, and this will no longer happen.

I'll leave this up to the Cura devs to decide if Cura should just throw an error when the user enters 0 as the minimum speed, or CuraEngine should be able to handle a minimum speeds of 0.

GregValiant, 2023-04-25:
Thank you @fieldOfView .

GregValiant, 2024-06-22:
The current version of Cura sets the {cool_min_speed} "minimum_value" to 1.0.  It would appear that the chances of this bug occurring again again are gone.
Can we close this?

closed 2024-07-07 by github-actions[bot]