Ultimaker/Cura #15343: Custom Printers use RepRap not Marlin.  Why will this setting not register?

TH-AC, 2023-04-24:
### Application Version

5.3.0

### Platform

Win 11

### Printer

Ender 3 custom Duet 3d and other custom Duet driven printer

### Reproduction steps

1 Select "Manage Printers"
2.  Select "Machine Settings" for any of my 2 printers
3. Change "G-code-flavor" to "RepRap
4.  Close the Machine settings page/Dialogue
5. Reopen Machine settings to confirm change of Flavor.  Flavor not changed still has Marlin as flavor.

### Actual results

Cannot change flavor to correct flavour

### Expected results

Change Flavor to RepRap.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Should there be a save button on the Machine settings page, for storing changes?

GregValiant, 2023-04-25:
In Explorer - right click on the "Cura 5.3" installation folder.  Select Properties.
On the Security tab "Edit" the permissions and make sure everything/everybody has access.
On the Sharing tab share the folder.
It's a Windows thing.  This is my computer, I'm the only one who uses it, and I have to share things with myself.  Dumb.
Anyway, those two things should allow you access and over-write ability.

GregValiant, 2024-06-22:
This bug has been addressed in current versions of Cura.  I'll go ahead and close this.

closed 2024-06-22 by GregValiant