Ultimaker/Cura #15347: Open With Ultimaker-Cura not working in File Explorer Windows 11 (Cura 5.3.0)

sumone-lol, 2023-04-24:
### Application Version

5.3.0

### Platform

Windows 11, Nvidia Geforce RTX 3050 Laptop GPU

### Printer

Creality Ender 3 Pro

### Reproduction steps

1. Open File Explorer
2. Navigate to .gcode or .stl file
3. Right click the file and select "Open with..."
4. Select Ultimaker-Cura.exe and click "Just once" or "Always" (it doesn't matter)

### Actual results

"Open with" window briefly closes and reopens and Cura doesn't launch, and now the "Always" option is missing

### Expected results

"Open with" window closes and Cura launches and opens the file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I've noticed that in the "Open with" window, Cura is called Ultimaker-Cura.exe and not UltiMaker-Cura.exe with a slightly different capitalization which I'm assuming is from a previous version since I've updated Cura a few times before. I tried choosing another app from my pc and selecting the actual Cura app, the list doesn't change the the problem reoccurs. I heard this is might be a Windows problem but this problem is unique to Cura and doesn't happen with other apps.
[image]
[image]

GregValiant, 2023-04-25:
When I first got this Win11 machine I had to set up the default programs for gcode files and STL files.  This is what worked:

- Right click on the file, select Properties.
- On the General tab there is a line for "Opens With".  Click on the Change button.
- If you can't find the app you want then keep moving to the bottom and eventually you come to "look for an app on this PC".
- Navigate to the Cura installation folder and find "Ultimaker-Cura.exe".

The relationship between files and what app they should open with is really a function of the operating system rather than the app itself so I don't think this is a Cura problem.  I'm inclined to remove the bug label and close this but I'll leave it open for a bit so you can respond here.  Let us know if you think that is premature.

sumone-lol, 2023-04-25:
I've tried that before but the part where it fails is the "Open with" menu which still appears in your method. I'll try some more drastic measures like deleting and reinstalling everything and possibly even looking through the registry editor. Honestly this might be just a me problem or a very small edge case.

GregValiant, 2023-04-25:
I think it's an edge case as well, but it's one that I seem to have had and I've read of others experiencing the problem as well so you aren't alone.

sumone-lol, 2023-04-25:
I tried deleting and reinstalling Cura and when I went to delete the old version, I found that the system still recognised Cura 5.2.1 as an app even though all of its files don't exist anymore. I deleted that too but it didn't fix the problem. Strangely, the Cura icon for .stl files returned which I'm assuming gets set as the default automatically but not for .gcode though. Out of curiosity, I renamed the UltiMaker-Cura.exe file to something else and then chose it to open a gcode file and it actually worked. The name from the previous version remained in the menu though so if I renamed it back, it would revert the change. At this point I'm pretty confident it's a Windows problem now.

sumone-lol, 2023-04-27:
So this was really weird and random. I went into the registry editor and went to `HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Explorer\FileExts`. I then found a file type that has already been set to open with Cura automatically like .stl or .3mf and basically just copied its contents to the .gcode folder to try to disassociate .gcode files and the non-existent application. Surprisingly, it actually worked. But version 5.3.1 was out and me being me, I updated it. Everything went pretty smoothly and I was able to fully uninstall the old version. However, Cura wasn't the default for .gcode files anymore but the correct one showed up in the "Open with" menu. I'm not sure which set of factors caused this behaviour, whether it was the registry editor or something about the new update but it's pretty much fixed for me.

closed 2023-04-27 by sumone-lol

samster395, 2023-07-22:
I am still having this issue, when I first updated cura I had multiple instances of cura in the open with menu, I managed to clear that and I now only have the one, current version but it still wont launch, just keeps opening the open with menu and when i click cura it goes away then comes straight back.

GregValiant, 2023-07-22:
@samster395 is that a Windows system?  The executable file for Cura was "Cura.exe" in the 4.x versions.  The 5.x versions changed to "UltiMaker-Cura.exe".  Is that the file that won't open from the right click menu?
As sumone-lol pointed out it could be a registry issue.

On my Win10 machine it is in "C:\Program Files\UltiMaker Cura 5.4.0".  Can you create a desktop shortcut that points to it?

samster395, 2023-07-22:
> @samster395 is that a Windows system? The executable file for Cura was "Cura.exe" in the 4.x versions. The 5.x versions changed to "UltiMaker-Cura.exe". Is that the file that won't open from the [...]

I'm on Win 11, it is the correct file name and yes I can create a shortcut to it.
Below is what the menu that opens looks like, clicking on just once just makes the same menu come back after a second, without actually opening Cura.

[image]

sumone-lol, 2023-07-23:
My temporary solution to this problem is to just rename the exe file to something else. It's kinda dirty but it works at least.

GoguyT3D, 2023-08-01:
> My temporary solution to this problem is to just rename the exe file to something else. It's kinda dirty but it works at least.

Im going to try this as I have a single monitor, 1050TI Nvidia Geforce, windows 11 and been having these open 3mf (with what?) problems quite a bit too.

GoguyT3D, 2023-08-02:
@sumone-lol I may have a resolution for you if you havent found one already, please bear with me as I have been having the open 3mf issue on and off for many cura versions. After reading all of everyones previous comments that got me to thinking. Windows is nutorious for screwing with app permissions and requiring you to "run as administrator" as it has for me in the past, so I went into my Cura 5.4 folder, right clicked on the Ultimaker-Cura.exe file, selected properties, and noticed windows had the permissions (under security tab) almost totally empty, I gave Ultimaker-Cura.exe full permissions (excluding special permissions) then clicked apply and ok. Now bear in mind this is all after I right clicked on my *.3mf that I wanted to open, selected "open with" then chose "choose an app from my PC" dug into the Ultimaker Cura 5.4 folder selected Ultimaker-Cura.exe and selected open by default. This seemed to have worked for me when shutting all apps down and trying to open a 3mf, I will restart my pc and try again to see if it its squashed. Thanks @samster395 @GregValiant 
[image]
[image]
another thing worth mentioning is that in my picture #2 you can see that windows has a compatibility option for use with you app in another version of windows, that may be an option 2 if this doesn't work

GoguyT3D, 2023-08-02:
Im back after a restart, now the 3mf opens properly, the window for open as a project displays properly the items are placed on the bed in the correct spot and when I clicked preview to see sliced lines the program crashed displaying (cura has stopped working) so that also made me think to go straight to the cura engine in the Ultimaker Cura 5.4 folder and also give that full permissions, that too has squashed the second bug (crashing) hope this helps others

GregValiant, 2023-08-02:
When I install a new version of Cura I have to copy in my custom files.  Windows won't let me do it until I go through the permissions and sharing thing.  So I can understand that GoguyT3D's should will work.

Guillaume337, 2023-09-16:
What worked for me:

Open regedit and go to Ordinateur\HKEY_CLASSES_ROOT\Applications\UltiMaker-Cura.exe\shell\open\command

then correct the value for the good folder.

"C:\Program Files\UltiMaker Cura 5.4.0\UltiMaker-Cura.exe" "%1"

It was still pointing old version folder

samster395, 2023-09-16:
> What worked for me:
> 
> Open regedit and go to Ordinateur\HKEY_CLASSES_ROOT\Applications\UltiMaker-Cura.exe\shell\open\command
> 
> then correct the value for the good folder.
> 
> "C:\Program Files\UltiMaker Cura 5.4.0\UltiMaker-Cura.exe" "%1"
> 
> It was still pointing old version folder

Thank you for this, it fixed it for me.