Ultimaker/Cura #15348: Benchy Roof collapsing using Cura 5.3.1 - Missing Layer # 159

Ergonomicmike, 2023-04-24:
### Application Version

5.3.1

### Platform

Win 10

### Printer

Ender 3 v2 with mriscoc firmware

### Reproduction steps

I tried to print a Benchy per the attached project file.
[3DBenchy.Roof Collapse problem.zip](https://github.com/Ultimaker/Cura/files/11314649/3DBenchy.Roof.Collapse.problem.zip)

### Actual results

At about layer 159 on the slicer-view preview (where the arch is about to close), the print head crashes downward onto the print.

### Expected results

I expected a Benchy like all the other times I've printed a Benchy. The last time I printed a Bench was with Cura 5.3.0. Although not quite the same print, because this time I switched to my 0.6 mm CHT nozzle and printed at a layer width of 0.4 mm. (As opposed to previously printing at 0.5 mm layer width with a 0.4 mm nozzle in Cura 5.3.0.)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

There is no layer 159 in the gcode.

I have played the gocde on various players/inspectors. (E.g. prusa-gcodeviewer.) I cannot duplicate the error when animating the gcode in the viewers. Although with one of the on-line viewers, there is a HUGE cylinder cross section showing near slicer-view layer 159.

Also, I used WinMerge to compare the gcodes from the same project file sliced in Cura 5.3.1 and 5.3.0. To my surprise, the gcodes are very different at places. Although neither show a layer 159 in the gcode.

GregValiant, 2023-04-24:
Thanks for the report.
I didn't have any trouble slicing it and I don't see any layers missing.  There are some errors within the STL (I downloaded one with errors in it as well).  After repairing the model it still sliced OK so I'm unable to duplicate the problem.
After saving the gcode I went through it and all the Z moves keep going upwards.  There are no moves that would cause the head to go downward and crash the print.
I'll leave the bug label on this for now, but you might have something else going on with your printer.  Are you printing from the SD card?

Ergonomicmike, 2023-04-24:
Thanks for the quick check and reply. (I always feel stupid when someone can't replicate my problem.)

Hmmm... now when I check for Layer:159 it's here in my gcode. So I don't know what happened before.

Interesting that the original Benchy has errors. (Which is what I assume you're saying.) 

Yes, I'm printing from the SD card. It will reformat it to make sure that it's not corrupt. (It passes chkdsk okay. But I'll reformat anyway.)

I'll close this for now.

closed 2023-04-24 by Ergonomicmike

GregValiant, 2023-04-25:
Computers are not an exact science.
I check every model for errors as that is far and away the #1 reason for problems.
I check all my Cura settings every time because for me, that is the #2 reason for problems as I forget I had something set and it's incorrect for the current model.  Then I still end up aborting because I forgot to move the Z seam or something.
#3 (and far down the list) are actual bugs but 5.2 had a lot of slicing problems with certain geometries and although it was worked on for 5.3.  I don't think it's been fixed yet.
At any rate, good luck with it.