Ultimaker/Cura #15350: An STL file created with OpenSCAD is not Splicing on version 5.3.0 and 5.3.1 but the same file is Splicing in 5.0.0 and 5.2.2. I have tried in both Windows Version and Mac Version, both has same issue. When Splicing it stops at around 30%

bala61, 2023-04-25:
### Application Version

5.3.0

### Platform

Windows 10 and Mac OS 13.3.1

### Printer

Creality Ender 3

### Reproduction steps

An STL file created with OpenSCAD is not Splicing on version 5.3.0 and 5.3.1 but the same file is Splicing in 5.0.0 and 5.2.2. I have tried in both Windows Version and Mac Version, both has same issue. When Splicing it stops at around 30%

[ventilation_grid_V3 - UmaMahesh 114.5.zip](https://github.com/Ultimaker/Cura/files/11316463/ventilation_grid_V3.-.UmaMahesh.114.5.zip)

### Actual results

The Splicing stops at around 30%

### Expected results

The Splicing stops at around 30%

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I have included the stl file above

nallath, 2023-04-25:
Thanks for reporting this! We're aware of an issue that causes some models to not slice. We don't have a fix yet, but we are investigating it.

In some cases it can help to move / rotate the model a bit. Although it's not a real fix, you might try this workaround for the moment

bala61, 2023-10-29:
Hi,

I have downloaded and installed Cura 5.5.0 on by Mac Studio and on Windows
PC.

As reported earlier, some files are not splicing with the latest version of
Cura as reported earlier for ver.5.3, whereas the same file is splicing
well on the old Cura version 5.2.2 on both MacStudio and Windows 10. The
same file is also splicing well in Prusa and Simplyfy3d.

I am enclosing the test STL file created using OpenScad for your reference.

Please do the needful.

Thanks

With warm regards
*U.Balashanmugham*  (Anand)

*U.Balashanmugham*
*Mobile 1: *+91 98410 26263
<http://voice.google.com/calls?a=nc,%2B919841026263>(WhatsApp)
*Mobile 2: *+91 94440 26263
<http://voice.google.com/calls?a=nc,%2B919444026263>
*E-mail: * ***@***.***>***@***.***
***@***.***>*

“*Tamil Arasu Illam*”

*Umapathy Lane, (Off Suncity Lane near Bata Showroom)#60 (old #16) Harleys
Road,Kilpauk,Chennai – 600 010India.*