Ultimaker/Cura #15354: Interleaving layers on a single model with support printing with second extruder.

Shlykov1, 2023-04-25:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Custom fdm

### Reproduction steps

1. Have a dual extruder printer
2. Load a single model with supports
3. Configure supports for the second extruder
4. Slice it

### Actual results

Interleaving layers on a single model with supports.

### Expected results

All layers should be with the same height.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

> Figure 1.

[image]
[image]

> Figure 2

[image]

Cura cannot slice correctly with 2 extruders when there is one model with support on the print bed, if there are more than one model, then everything prints correctly. The essence of the problem is that cura alternates layers, as shown in Figure 1, one layer is normal, and the second is reduced. I also opened the gcode sliced with cura in an ideamaker 3d and it shows that the layers alternate in the 
same way as on the presented models in Figure 2.
[cura_bug.zip](https://github.com/Ultimaker/Cura/files/11321393/cura_bug.zip)

nallath, 2023-04-25:
Thanks for the extensive issue. This helps a lot with debugging this!

I do see from your logs that you have a number of plugins that might influence the Z height, which could be the cause of this. Could you confirm for me if they are enabled for this print? 

If that's not the issue, I believe that it is probably caused by the raft. My rationale for that is that most of the Ultimaker profiles don't use rafts and the case that you describe is something that we tested with our profiles (but without a raft). It could well be that the "weird" layerheights that a raft can have are not correctly taken into consideration for support, causing the interlaced affect that you are seeing.

Shlykov1, 2023-04-25:
Thanks for fast answer! I have installed z_offset plugin, but wasnt used it. If we consider on this print, there is only a "thumb nail" plugin.

nallath, 2023-04-25:
In that case, it's likely the raft that is the culprit here. I will discuss it with the rest of the team!

Shlykov1, 2023-04-25:
Thanks a lot, i'll be looking forward to the new patch!

Shlykov1, 2023-04-26:
Hello! I found the problem, it was not on your side. Apparently, I have a backlash in the screw-nut pair that moves the table up and down, when the support extruder went down, I had to lower the table lower than in the previous layer, and then raise it back.

MariMakes, 2023-05-08:
Hey @Shlykov1,

Thanks for the update 👍

Since it's not a bug in Ultimaker Cura anymore. 
I'll be closing this issue. 

Happy printing 😄

closed 2023-05-08 by MariMakes