Ultimaker/Cura #15356: Slicing Failure Bug

Regner53, 2023-04-25:
### Application Version

5.2.2

### Platform

Windows 10

### Printer

Anet A8 Plus

### Reproduction steps

Just slicing a large part with supports and it crashes.

### Actual results

It crashes after a few seconds and suggests I report a bug to you

### Expected results

No crashy crashy

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11323617/cura.log)
[AA8P_Tankard3.zip](https://github.com/Ultimaker/Cura/files/11323650/AA8P_Tankard3.zip)

nallath, 2023-04-25:
Thanks for reporting this! We're aware of an issue that causes some models to not slice. We don't have a fix yet, but we are investigating it.

In some cases it can help to move / rotate the model a bit. Although it's not a real fix, you might try this workaround for the moment

SogenCookie, 2023-04-25:
I am also having this problem on 5.3.0

Regner53, 2023-04-26:
Thank you for your response.  Rotating the part about 45 degrees solved the
problem.