Ultimaker/Cura #15360: Bug. Support blockers?

nitram6272, 2023-04-26:
### Application Version

5.3.0-alpha+xmas

### Platform

Windows 10

### Printer

Artillery Sidewinder X1

### Reproduction steps

Used cubed support blockers on 2 of 3 models for the first time.

### Actual results

Failed to slice approx 60% into slicing.
[image]

### Expected results

Sliciing should complete.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11330083/cura.log)

nitram6272, 2023-04-26:
Although I created a project file it would not upload sorry.

GregValiant, 2023-04-26:
Thanks for the report.
Github site rules are that files other than text files must be zipped in order to upload.  Right-click on the 3mf file and select "Send To..." and then "compressed (zip) folder".  Post the zip folder here.

closed 2023-05-30 by github-actions[bot]