Ultimaker/Cura #15361: units overlap text

AMI3, 2023-04-26:
### Application Version

5.3.1

### Platform

windows 11

### Printer

creality ender3

### Reproduction steps

  I am using windows in hebrew, and it seems the Right to Left indentation messes up the units position, and overlaps the input text in the configuration menu
[image]

[image]

### Actual results

The units position, and overlaps the Data text.
[image]

### Expected results

No overlap of the text

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I added 2 screensots
[image]
[image]

MariMakes, 2023-05-08:
Hey @AMI3,

Sorry, it took us a while to get back to you 😞 

Thanks for your report 👍 This is the first time I've seen this. 😅 

I'll bring it up to the team to see what they can do to improve it. 
Fingers crossed 🤞

MariMakes, 2023-08-07:
Hey @AMI3,

Quick update from our side 👋

One of our developers was able to resolve it, so it should be fixed in the next release 😊
You can read more about the fix here https://github.com/Ultimaker/Cura/pull/16270

Thanks again for your report 👍

closed 2023-08-07 by MariMakes