Ultimaker/Cura #15362: Problem with slicing

spideyguy-one, 2023-04-26:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Anycubic Mega Pro

### Reproduction steps

Loaded the articulated dragon STL, then pressed Slice, got Slicing Failed error. Previously, when I used the Alpha Xmas version of Cura, I sliced and successfully printed the dragon. I am unable to do so with the new version.
When I load the dragon the Mesh tools pops up a message that the model is not watertight, but it still printed the first time.
[image]

### Actual results

I get the Slicing Failed message.

### Expected results

It should complete slicing and allow me to print the model.
[CrystalDragon_NewLegs_Curled_2(1).zip](https://github.com/Ultimaker/Cura/files/11333426/CrystalDragon_NewLegs_Curled_2.1.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

All items included

GregValiant, 2023-04-26:
Thanks for the report.
The Crystal Dragon model has a couple of different versions.  The one you posted has errors.  This is a screenshot from Mesh Mixer.  The red and magenta balls indicate errors in the model.  A portion of the tale is all magenta so it likely has a lot of inverted surfaces.
[image]

Mesh Mixer was not able to repair the model.  I tried rotating your model on the Cura build plate and it sliced.
The Cura Team will take a look at this.
[image]

spideyguy-one, 2023-04-26:
Thank you for the prompt reply. I found that I had no problem slicing 
that model with version 5.1, so I renamed my AppData\Roaming\cura\5.1 
folder to 5.3, restarted Cura, and was able to slice the model. It works 
great now.
Oddly, putting back the original 5.3 folder gives the slicing error for 
all of the dragon models I tried to slice. Replacing it with the 5.1 
folder contents lets me slice them with no problems.

Thanks again!

image
<https://user-images.githubusercontent.com/64202104/234605745-f70d2040-14e2-4775-be70-066e601bb00a.png>

image
<https://user-images.githubusercontent.com/64202104/234608243-f7477d35-501c-4c3e-a45a-893ca4fecdc6.png>