Ultimaker/Cura #15364: Cura UI is not showing up

sanyi, 2023-04-26:
### Application Version

5.3.1

### Platform

XUbuntu 22.04

### Printer

custom / does not matter

### Reproduction steps

Start the apk 5.3.0 or 5.3.1 (does not matter which libux/linux modern).

### Actual results

After the loading part the UI disappear while the application remains loaded.

```
ps aux | grep Cura
attila    208832  0.3  0.0  12348  2564 ?        Ssl  20:03   0:01 /home/attila/utils/UltiMaker-Cura-5.3.0-linux-modern.AppImage
attila    208837  1.0  0.4 2947240 276428 ?      Sl   20:03   0:06 /tmp/.mount_UltiMa3X2DIN/UltiMaker-Cura
attila    208903  0.0  0.0 144716  5588 ?        SNl  20:03   0:00 /tmp/.mount_UltiMa3X2DIN/CuraEngine connect 127.0.0.1:49674
attila    214939  0.0  0.0   9208  2244 pts/0    S+   20:13   0:00 grep --color=auto Cura
```

in the logs i can see 
```
2023-04-26 20:03:24,410 - DEBUG - [MainThread] UM.Application.initialize [163]: Initializing UltiMaker Cura
2023-04-26 20:03:24,410 - DEBUG - [MainThread] UM.Application.initialize [164]: App Version 5.3.0
2023-04-26 20:03:24,410 - DEBUG - [MainThread] UM.Application.initialize [165]: Api Version 8.3.0
2023-04-26 20:03:24,410 - DEBUG - [MainThread] UM.Application.initialize [166]: Build type None
2023-04-26 20:03:24,411 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib/uranium must be a folder.
2023-04-26 20:03:24,411 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib64/uranium must be a folder.
2023-04-26 20:03:24,411 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib32/uranium must be a folder.
2023-04-26 20:03:24,411 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/.mount_UltiMa3X2DIN/plugins must be a folder.
2023-04-26 20:03:24,411 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/Resources/uranium/plugins must be a folder.
2023-04-26 20:03:24,411 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/Resources/cura/plugins must be a folder.
2023-04-26 20:03:24,411 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [481]: Initializing storage paths
2023-04-26 20:03:24,411 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [491]: Config storage path is /home/attila/.config/cura/5.3
2023-04-26 20:03:24,411 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [499]: Data storage path is /home/attila/.local/share/cura/5.3
2023-04-26 20:03:24,411 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [511]: Cache storage path is /home/attila/.cache/cura/5.3
2023-04-26 20:03:24,411 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /tmp/.mount_UltiMa3X2DIN/share/uranium/resources/bundled_packages/uranium.json
2023-04-26 20:03:24,411 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /tmp/.mount_UltiMa3X2DIN/share/cura/resources/bundled_packages/cura.json
2023-04-26 20:03:24,411 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [124]: Trying OpenGL context 4.1...
2023-04-26 20:03:24,411 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [132]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
2023-04-26 20:03:24,411 - DEBUG - [MainThread] UM.Qt.QtApplication.initialize [192]: Detected most suitable OpenGL context version: 4.1 Core profile
2023-04-26 20:03:24,411 - INFO - [MainThread] UM.Qt.QtApplication.initialize [199]: Initializing job queue ...
2023-04-26 20:03:24,411 - INFO - [MainThread] UM.Qt.QtApplication.initialize [203]: Initializing version upgrade manager ...
2023-04-26 20:03:24,411 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from /tmp/.mount_UltiMa3X2DIN/share/uranium/resources/bundled_packages/uranium.json
2023-04-26 20:03:24,411 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from /tmp/.mount_UltiMa3X2DIN/share/cura/resources/bundled_packages/cura.json
2023-04-26 20:03:24,411 - INFO - [MainThread] UM.PackageManager._loadManagementData [201]: Loaded user packages management file from /home/attila/.local/share/cura/5.3/packages.json
2023-04-26 20:03:24,412 - INFO - [MainThread] UM.PackageManager._removeAllScheduledPackages [280]: Attempting to remove the following scheduled packages: - 
2023-04-26 20:03:24,412 - INFO - [MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [121]: Loading plugin configuration file '/home/attila/.config/cura/5.3/plugins.json'
2023-04-26 20:03:24,412 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib/cura must be a folder.
2023-04-26 20:03:24,412 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib32/cura must be a folder.
2023-04-26 20:03:24,412 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/libx32/cura must be a folder.
2023-04-26 20:03:24,412 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib64/cura must be a folder.
2023-04-26 20:03:24,412 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [447]: Plugin [SentryLogger] has been disabled. Skip loading it.
2023-04-26 20:03:24,412 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin FileLogger 1.0.1
2023-04-26 20:03:24,418 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin LocalContainerProvider 1.0.1
2023-04-26 20:03:24,424 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin LocalFileOutputDevice 1.0.1
2023-04-26 20:03:24,430 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin CameraTool 1.0.1
2023-04-26 20:03:24,435 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin MirrorTool 1.0.1
2023-04-26 20:03:24,443 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin RotateTool 1.0.1
2023-04-26 20:03:24,449 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin ScaleTool 1.0.1
2023-04-26 20:03:24,453 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin SelectionTool 1.0.1
2023-04-26 20:03:24,459 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin TranslateTool 1.0.1
[... 143 lines cut ...]
2023-04-26 20:03:27,936 - DEBUG - [MainThread] ModelChecker.ModelChecker._createView [118]: Creating model checker view.
2023-04-26 20:03:27,946 - DEBUG - [MainThread] ModelChecker.ModelChecker._createView [127]: Model checker view created.
2023-04-26 20:03:27,946 - DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._createDialog [76]: Creating dialog [MoreInfoWindow.qml]
2023-04-26 20:03:28,410 - DEBUG - [MainThread] cura.CuraApplication.run [884]: Booting Cura took 4.394691705703735 seconds
2023-04-26 20:03:28,411 - DEBUG - [JobQueueWorker [0]] cura.Machines.ContainerTree.run [174]: Started background loading of MachineNodes
2023-04-26 20:03:28,622 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [221]: Error check finished, result = False, time = 1.59s
2023-04-26 20:03:29,363 - DEBUG - [JobQueueWorker [0]] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for creality_ender3pro took 0.45096421241760254 seconds.
2023-04-26 20:03:30,544 - DEBUG - [JobQueueWorker [0]] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for creality_ender3 took 0.17971515655517578 seconds.
2023-04-26 20:03:31,045 - DEBUG - [JobQueueWorker [0]] cura.Machines.ContainerTree.run [185]: All MachineNode loading completed
```

Note that 5.2.2 works.

### Expected results

application should be visible.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

fieldOfView, 2023-04-28:
This is caused by the MeshTools plugin version 3.7.1. You have to either uninstall that plugin or update it to version 3.7.2.
See this post for instructions on how to do this: https://github.com/fieldOfView/Cura-MeshTools/issues/36

fieldOfView, 2023-04-28:
Since this is not an issue caused by Cura itself, but a plugin, and since there is a fix, I have closed this issue on the Cura repository.