Ultimaker/Cura #15366: Crash during slicing using tree supports

tommyboyiceman, 2023-04-27:
### Application Version

5.3

### Platform

Windows 11

### Printer

Anycubic Kobra Max

### Reproduction steps

Attempting to slice a large object using tree supports causes error. Works properly when using grid supports.

### Actual results

Error and did not slice

### Expected results

Sliced with tree supports

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11339160/cura.log)
[AKMAX_Bottom For Drone and400mmtall Bed.zip](https://github.com/Ultimaker/Cura/files/11339167/AKMAX_Bottom.For.Drone.and400mmtall.Bed.zip)

ScienceofSpock, 2023-05-03:
Wow, This is incredibly similar to the issue I am running into, except I am on Win 10 Pro instead of Win 11, and my issue is on 5.3.1 (Though, I also had it on 5.3). I have a large model, ~390mm in the Y dimension. It slices just fine with "normal" supports, but when I try to slice it with tree supports, it fails. It slices for about 45 mins, with the progress bar slowly advancing, then fails with unexpected error.

I have attached my project file and log file for reference, instead of opening a new bug report.
[cura_log.zip](https://github.com/Ultimaker/Cura/files/11380207/cura_log.zip)
[AKMAX_Cabin Body_v6.zip](https://github.com/Ultimaker/Cura/files/11380208/AKMAX_Cabin.Body_v6.zip)

ScienceofSpock, 2023-05-03:
I actually got it to slice last night by changing my changing my support overhang angle to 21, but that introduced a "crease" in the tree supports inside that starts unsupported at around layer 210. It's not possible that this gcode will work, it's trying to print on air. I can provide my gcode, but it's large. Even zipped, it's 33MB.

ScienceofSpock, 2023-05-03:
@tommyboyiceman it looks like your log is showing the same issues mine is. Right before it fails, it throws a few "Failing to discretize parabola! Must add an apex or one of the endpoints." and then "Backend exited abnormally with return code 3221225477!"

ScienceofSpock, 2023-05-03:
I managed to get it to slice a second time with tree supports, support overhang angle set to 45 degrees, 10 % infill, but again, it is trying to print support material in thin air. These appear to be caused by features in the floor of my model (Which is being printed upside down). I will remove those features and try to slice again.
[image]

MariMakes, 2023-05-08:
Hey all,

Thanks for the reports 👍
Sorry, it took us a while to get back to you 😞 

@ScienceofSpock We are working hard to improve tree supports for Cura 5.4 because we want to include the new tree supports from Thomas Rahm. We've already made significant improvements since the release of the Alpha here: https://github.com/Ultimaker/Cura/discussions/14111
I checked your file on a recent build, it  slices successfully and looks like this: 
[image]
[image]
You'll have to wait for a next release to see it in action. 

@tommyboyiceman we are still missing your project file 
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

tommyboyiceman, 2023-05-08:
Added my project file to my original post

ScienceofSpock, 2023-05-08:
@MariMakes Those supports look great, and I look forward to using them in the future. My 5+ day print with standard supports is going to finish in an hour or three :)

MariMakes, 2023-05-09:
Hey @tommyboyiceman,

Thanks for the project file👍
I can reproduce your crash on my side, even in a recent build.
[image]

I did notice that tilting the model a little resulted in a successful slice. I hope that helps you get unstuck. 
[image]

I'll have someone from the team take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0