Ultimaker/Cura #15367: Slicing in Cura 5.3.1

eSchale1950, 2023-04-27:
### Application Version

5.3.1

### Platform

Windows 11

### Printer

Artillery Sidewinder X1V4

### Reproduction steps

STL-Datei wird in Cura 5.3.1 geladen und ist Sichtbar, verschwindet aber nach kurzer Zeit.
Kein Slicen es wird 0 Minuten angezeigt, man muss unter Datei alle Modelle neu wählen, das Teil ist wieder Sichtbar dann findet das Slicen statt und die Tauer des Druckauftrages wird angezeigt.
In Cura 5.2.2 ist das nicht der Fall.

### Actual results

In Cura 5.3.1 nichts

### Expected results

Programm soll so laufen wie Cura 5.2.1

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Bitte um Info

GregValiant, 2023-04-28:
Thanks for the report.
I am running Windows 11 and Cura 5.3.1 and I have not experienced the problem.  Use the "Help | Show Configuration Folder" command to locate the Cura.log file.  Post the log file here.

eSchale1950, 2023-04-30:
Als Datei Cura Log
[cura.log](https://github.com/Ultimaker/Cura/files/11362081/cura.log)

MariMakes, 2023-05-08:
Hey @eSchale1950,

Entschuldigung, dass es nicht auf Deutsch ist, aber das könnte helfen.

I've also never seen it before. 
Does this happen with a specific model?
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file.

Do you have a screenshot or even better a screen recording for us?
That would really help with troubleshooting. 😇

closed 2023-05-29 by github-actions[bot]