Ultimaker/Cura #15368: Support layer height

calculuschild, 2023-04-27:
### Is your feature request related to a problem?

Support printing takes a significant portion of print time, particularly tree supports.

Further, when working at fine detail (0.1 mm layer height or smaller) the thin walls of isolated tree branches tend to gradually get nudged out of position with so many fine passes.

### Describe the solution you'd like

Similar to the option for different infill layer heights, I'd like a way to set separate support layer heights. My tree supports don't need the fine detail of the rest of my model.

### Describe alternatives you've considered

Printing at a thicker layer height seems to generate more reliable supports (and is much faster). But then I lose out on detail on the actual part.

### Affected users and/or printers

Users who print with fine detail and use lots of supports (miniatures printing, etc). But most people should benefit from some kind of speedup.

### Additional information & file uploads

Seems like it would be fairly simple to implement, since this is already an option for infill.

GregValiant, 2023-04-27:
Currently there is no setting for the Support Wall Layer Thickness.  They go down every layer.  There is a setting for "Support Infill Layer Thickness".  I generally set it to 2X layer height.  That works OK for larger support structures or higher Support Infill Density.  Other than that it doesn't have much of an effect.  It is also bound to "Flow Rate Compensation Factor".  When you are flowing twice as much material Cura will slow down the print speed to account for it.  At 0% there is no slow-down to compensate for the increased layer height.

calculuschild, 2023-04-27:
> Currently there is no setting for the Support Wall Layer Thickness. 

Yep, you got it. That's exactly the feature I'm requesting be added. :+1: Sounds like we are on the same page.

johnmatthews7234, 2023-05-02:
In addition, a similar setup for inner wall would be a nice to have...  I don't need the quality of a 0.04mm layer height for an inner wall, but 0.04 is fantastic for highly detailed models.  

It's a "nice to have..."