Ultimaker/Cura #15370: Editing start and end G-codes

teho03, 2023-04-27:
### Application Version

5.3.1

### Platform

Win10 64 bit

### Printer

Ender 3 V2

### Reproduction steps

Going to the Preferences/Printers/Machine settings I found that as soon as I moved the cursor into either the start or end G-code regions the 'help screen' appears over the top of the area making it impossible to edit the G-codes. 

### Actual results

Cannot edit start/end G-codes

### Expected results

Am able to edit start/end G-codes

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fieldOfView, 2023-04-27:
The Settings Guide plugin makes the tooltip much bigger than envisioned. You can fix this by disabling the plugin, or disabling its option to put the articles in tooltips.

MariMakes, 2023-05-08:
Hey @Gerhard-59,

Welcome to the Ultimaker Cura Github 🚀

This is related to the Settings Guide Plug-in.
People who don't have that plug-in installed, like me, just see this as tooltip:
[image]

You can temporarily disable your plug-in in the marketplace by clicking the gear in the top right corner and then disable.
If you are done editing your gcode you can enable it again.
[image]

It's not a bug in Ultimaker Cura but one in the Settings Guide Plug-in.
I'll be closing this issue. I hope you understand.

closed 2023-05-08 by MariMakes