Ultimaker/Cura #15372: Adaptive Layer Height Behavior

Sineos, 2023-04-27:
### Application Version

5.2.2, 5.3.1

### Platform

Win 10

### Printer

Custom

### Reproduction steps

Use Adaptive Layer Height with Tree Support

### Actual results

[image]

Yellowish tones: Layer-Height 0.3 to 0.35
Blue tones: Lower

As you can see from the screen-shot the ALH algo is making a poor choice:
Instead of slicing the supported overhang already with a finer LH it starts a good deal above. Since there is a thread in the hole, its start also gets printed with the high LH.

It can be improved with playing with the _ALH Maximum Variation_ setting but it keeps adamant on printing the first overhang layer with the high layer height.

### Expected results

Slicing the overhang with a fine LH would surely yield better results and less sagging. Apart from this the thread in the hole would print better as well

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[test.3mf.txt](https://github.com/Ultimaker/Cura/files/11346868/test.3mf.txt)

GregValiant, 2023-04-27:
Thanks for the report.
Could you switch printers in Cura and try another Project File?  Your "Custom" printer won't install for me and consequently I don't get your Cura setup.
With the layer height at .10 and the Support Z Distance at .10 I'm getting a single layer air gap over the support.
The "Support Z Distance" doesn't appear to update when I change the regular layer height.

Sineos, 2023-04-28:
> Could you switch printers in Cura and try another Project File? Your "Custom" printer won't install for me and consequently I don't get your Cura setup.

My pleasure: [V350_test.3mf.txt](https://github.com/Ultimaker/Cura/files/11350032/V350_test.3mf.txt)

> With the layer height at .10 and the Support Z Distance at .10 I'm getting a single layer air gap over the support.
> The "Support Z Distance" doesn't appear to update when I change the regular layer height.

In fact it is two issues:
* The first overhang layer should be already a thin layer. IMO forcing it to thick is detrimental
* You are right: The gap to the support is too high. Didn't even notice this initially. Thanks for spotting

GregValiant, 2023-04-28:
To make the first layer of the overhang a finer layer height the "Adaptive Layer Variation Step Size" must be equal to the "Adaptive Layer Maximum Variation".  That forces Cura to jump directly from Maximum Layer Height to Minimum Layer Height with no steps in between.
Here I have set the Layer Height to 0.20 and both of those Adaptive Settings to 0.10.  The minimum layer height is therefore 0.10 and the maximum is 0.30.  It isn't really "Adaptive" anymore, but from the first layer of the overhang to the top layer of the model is .10 layer height.  You can see that  Layer:0 is 0.20 layer height with a single step to 0.30 as the support and bottom of the model build upward.
[image]

In your image you can see that from the build plate up to "Maximum Layer Height" there are a few steps.  Then from "Maximum Layer Height" down to "Minimum Layer Height" (at the overhang) there are twice that many steps as Cura calculates from Max down through Layer Height and continues to Min.  The layer height is "stepping" down by "...Variation Step Size".  In mine it is doing the same thing, but there is a single step involved rather than multiple steps.

I don't really see a Cura bug here.  You have a very specific condition and it looks like the settings that are available can be made to fit the condition.

Off topic - it looks like your "tap drill" is on the big side.

Sineos, 2023-04-28:
Thanks. Your support is highly appreciated

closed 2023-04-28 by Sineos