Ultimaker/Cura #15373: Reduce popup message repitition.

tlhintoq, 2023-04-28:
### Application Version

5.3.1

### Platform

Windows10x64, 21H2

### Printer

All - Not printer specific

### Reproduction steps

With print sequence set to 'one at a time'.
With supports turned on
Add a model.
Now start adding blockers.
Every time you do anything... add... delete.. move the model.. You get a popup over an already crowded UI letting you know the print volume height has been reduced due to print sequence.

Ok.  we get it.  It doesn't have to bang with every little move or action.  Maybe just when that one setting is touched.
[image]

### Actual results

Popup is just continuous on every action when it doesn't need to be.

### Expected results

Just show popups related to print sequence, when the user changes print sequence.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2023-05-05:
Hey @tlhintoq,

Thanks for your report 👍
I don't print a lot of One at a time, but I can imagine that can annoying. 😅 

I'll bring it up to the team to see what they can do to improve it. 
But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞

closed 2023-05-08 by nallath

MariMakes, 2023-05-08:
Hey @tlhintoq,

Quick update on our side. 
One of our Developers took a look and figured out what was wrong. 
It should be resolved in the next release.

Thanks for the report 👍

tlhintoq, 2023-05-08:
Great news.  Thanks for the update

commit 3f837992 (2023-06-13): Only show build volume height message if it was changed

Fixes #15373 && CURA-10580