Ultimaker/Cura #15374: découpe avec support normaux

fab7294, 2023-04-28:
### Application Version

5.3.1

### Platform

mac os ventura 13.2

### Printer

ender 3

### Reproduction steps

demande de tranchage de la pièce en position vertical avec supports normaux, cura bug et ne veut pas tranché.

### Actual results

tranchage impossible avec mes paramètres.

### Expected results

le pièce devrait être tranché et la visualisation des couches devrai être possible.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]

fab7294, 2023-04-28:
[image]

MariMakes, 2023-05-08:
Hey @fab7294,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Have you tried reverting to the default settings?
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-05-29 by github-actions[bot]