Ultimaker/Cura #15381: Custom Wall Printing Order

TIMMOOOH, 2023-04-28:
### Is your feature request related to a problem?

Now there are only two options for the order, walls are printed in. For me, when printing from inside to outside, the layers look shifted and not even, because the wall next to the outer wall restricts filament flow and pushes it to the outside. When printing from outside to inside, the walls look great and even, but the start of the outer layer is underextruded because of potentially long travels. "Retraction extra prime ammount" doesn't always fix the issue as the ammount of filament missing at the start is depending on the travel move to the outer wall.

### Describe the solution you'd like

A option for custom wall ordering would be nice, perhaps in these brackets  "[ ]":
(3 walls in this example)
1 = outer wall
2 = wall between 1 and 3 
3 = inner wall next to infill

The setting would look like this:
Custom Wall Printing Order            [3 ,1, 2]

In this example the printer would start with the innermost wall next to the infill, where the ugly start can't be seen from outside and than print the outer wall, when the nozzle is primed, and then the middle.

### Describe alternatives you've considered

A third option next to "inside to outside" and "outside to inside" called "inside outside middle" or something where it first prints the third wall from the outside, then the first, then the rest. like this: (Example with 4 walls)

outside
-------------------2.
-------------------3.
-------------------1.
-------------------4.
inside
infill

### Affected users and/or printers

everyone who uses "outside to inside"

### Additional information & file uploads

I am aware of #12928 and #11934. I don't think this is a duplicate of #11934 as my issue has nothing to do with a center line.

GregValiant, 2023-04-28:
Have you tried the post processor "Retract Continue"?  It stretches the retraction across the following travel moves and keeps oozing down.  The nozzle pressure will come back up quicker on the prime.  Be advised that it doesn't like Z-hops as they confuse the post processor.

TIMMOOOH, 2023-04-28:
@GregValiant Thank you, i will try it with my next print. Does this override the normal retraction distance settings?