Ultimaker/Cura #15382: Sudden failure of slicing with custom profile

ericspivey, 2023-04-28:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

I tried to slice the file skullandblock-230428 using a custom profile "fast normal supported" (both in attached zip). It proceeds about halfway and then terminates with an "unexpected error"

### Actual results

It proceeds about halfway and then terminates with an "unexpected error"

### Expected results

I would expect it to create a .ufp file

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Here are the files in a zip.
[Bug Report Files.zip](https://github.com/Ultimaker/Cura/files/11356580/Bug.Report.Files.zip)

GregValiant, 2023-04-28:
Thanks for the report and the files.  I wasn't able to slice it either.
The screenshot below is from Mesh Mixer and it is not an acupuncture chart.  Each ball indicates an error in the mesh of the model.  It's a good habit to check files for errors before putting time into them.  
[image]

This is from an on-line site I use to check models.
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 72 Non-manifold edges (?)
--> 112 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 8 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 28895ms ------
-> Vertex count changed from 120421 to 114495 (-5926)
-> Triangle count changed from 243022 to 231418 (-11604)

I was able to repair the model using MS 3D Builder but the repaired model still wouldn't slice in Cura.  Prusaslicer finished slicing the repaired model but informed me there was an error in the toolpath.
Someone from the Cura team will take a look.

ericspivey, 2023-04-28:
Thanks so much for looking into this. I actually made the .stl by merging (boolean union) the meshes of the skull and the mount using MeshLab. I tried to remove redundant/close vertices, but this didn't do much. I should see if MeshLab will let me remove non-manifold edges and inverted faces, since that looks to be the biggest problem. Is there an accepted way to merge meshes that I have perhaps missed? I actually have a fair amount of flexibility in how this is done, and can accept a lower resolution product if needed.

GregValiant, 2023-04-29:
I don't think the resolution is a problem.  I switched printers to my Ender and it sliced.  This is the repaired model and the settings are my normal everyday profile.
[image]

So I'm thinking that although there are some errors that the errors can be overcome.  This may be a bug in Cura but it will be a while before that will be addressed.  In the mean time, play with the support settings.  Some of the "won't slice" bug reports here on GitHub seem to indicate that certain combinations of support settings aren't playing nice together.  I can't be any more specific than that but it's a feeling I have and it's something you could try now.  I know that adjusting the Initial Layer Height to an odd number (like 0.201mm) can get some stubborn models to slice.  Moving or rotating the model on the Cura build plate can sometimes get a model to slice.

Also (and this is just an aside) if you clean out that skulls sinuses and get rid of the stalactites hanging from the roof it might have a better chance as well.  Trying to slice and build internal support around delicate features like those is tough.  The slice took quite a while (upwards of 20 minutes) and I have a feeling that Cura found it difficult even though it snot.
(Sorry...I couldn't resist)

ericspivey, 2023-05-01:
Hi Greg, this is great to know -- I had been trying to introduce fewer variables by keeping a simple orientation, layer height, etc., but what you say about making things at least slightly off-kilter makes sense. For functional/mechanical reasons, I'll probably have to print at different orientations as well, so it shouldn't be a problem. I will try out these strategies and let you know how it goes.

I had some follow on questions
1. Is there any algorithm for CURA that minimizes NOT the volume of support material needed, but the area of build needing support?
2. Is there a way to exclude support/printing in a certain volume? We are considering adding non-printed elements during the print, and would like to mark the volume as "no-go"
3. You mentioned being able to slice with your Ender printer software. Is there "better" software for this sort of thing, including that which we'd have to pay a license for? I work in academic research, so this is a possibility, within reason, though I'd like to avoid buying another printer.

GregValiant, 2023-05-02:
**1. and 2.**  There are several ways to limit the areas that support is created for.  The "Support Overhang Angle" is # 1.  If you look at the underside of an overhang you will see it is red.  If you increase the "Support Overhang Angle" the red area will get smaller on something like a sphere (it won't on a flat surface because any angle is constant).  About the max you can go with the overhang angle is 63°.  Above that and the extrusions don't stick together.  Another setting is the Support XY distance.  A higher distance will move the support away from an adjacent wall.
On the left side toolbar you will find a button for Support Blockers.  When a support blocker is in contact with that red area (on the under side of a surface) then support will not be generated there.  The blocker does not have to hit the build plate, just the overhang you want to effect.  Blockers can be scaled in all three dimensions like any other mesh.  In addition, support blockers can be configured as Cutting Mesh and Infill Mesh and can have different properties.  The button above the support blocker button is "Per Model Settings".  I'll let you play with it as it can get quite involved.  In the Cura MarketPlace is a nice plugin for Cylindrical Supports.  They can be configured like a cube shaped support blocker.  Their default is "Print as Support" so by default rather than blocking they add support.  They can be changed by adjusting their individual settings.
3.  I try to keep up with PrusaSlicer, IdeaMaker and other slicers but I print with Cura.  There is a learning curve for all of them.  I'd rather know one well than three a little bit each.  Cura is also highly customizable which I like.

This is a shape I have for playing with supports.  There is a single blocker in place (it's has the blue outline on the left) and it covers just a bit of the underside overhang and the top of the horizontal hole.  You can see the other holes have support in them.  The angle on the front overhang is 45° but it doesn't get support because I've made the "Support Overhang Angle "55°.  I generally prefer normal supports but tree supports have their place.
[image]

At my last count Cura has 73 support settings.  Throw in the blockers and tree support structure and It's like it's own science trying to get them just right.  The first thing I want to know is "How do I divorce the print from the support without wrecking anything".  Hobby knives, "pics" that are like dental tools but are more robust, a dremel tool can come in handy, needle nose pliers, a 1/4" wide wood chisel, a set of micro-files....the list gets long.  Supports can be a real challenge.  Sometimes I've had to design them myself in CAD because I didn't like what the software was doing.

ericspivey, 2023-05-02:
This is great, thanks -- I have played with the angle and layer space settings, and haven't struck gold yet. I will try playing with overhangs. I did find some .ufp files of this or a similar part that a predecessor on the project generated, but I am a little lost as to how to mine it for information -- when I load it, the print settings are locked and are not displayed. I extracted the raw gcode from the .ufp and have tried to glean that for information, but have found little I didn't already know - is there a way to find out overhang from the gcode? There are these "setting" lines at the end:

;End of Gcode
;SETTING_3 {"global_quality": "[general]\\nversion = 4\\nname = Fast #2\\ndefini
;SETTING_3 tion = ultimaker_s5\\n\\n[metadata]\\ntype = quality_changes\\nqualit
;SETTING_3 y_type = draft\\nsetting_version = 20\\n\\n[values]\\nprime_tower_ena
;SETTING_3 ble = False\\nsupport_enable = True\\nsupport_extruder_nr = 1\\n\\n",
;SETTING_3  "extruder_quality": ["[general]\\nversion = 4\\nname = Fast #2\\ndef
;SETTING_3 inition = ultimaker_s5\\n\\n[metadata]\\ntype = quality_changes\\nqua
;SETTING_3 lity_type = draft\\nsetting_version = 20\\nposition = 0\\n\\n[values]
;SETTING_3 \\ninfill_pattern = lines\\ninfill_sparse_density = 100\\nwall_line_c
;SETTING_3 ount = 6\\n\\n", "[general]\\nversion = 4\\nname = Fast #2\\ndefiniti
;SETTING_3 on = ultimaker_s5\\n\\n[metadata]\\ntype = quality_changes\\nquality_
;SETTING_3 type = draft\\nsetting_version = 20\\nposition = 1\\n\\n[values]\\n\\
;SETTING_3 n"]}

Is this telling me that the infill is 100% with pattern "lines"?

ericspivey, 2023-05-03:
I figured out how to use the gcode to import settings back into Cura, and was able to slice one of the skulls using those settings, though I'm not sure how it's different from what I had been trying before - I think it's the wall settings? If there's a way for me to send you the gcode or ufp file, it may be useful for figuring out what's going on.