Ultimaker/Cura #15384: Slicing a body of 4 parts,helmet,head,backpack, of astronaut stl to print

JoelStolarski, 2023-04-29:
### Application Version

5.3.1 Stable

### Platform

Windows 10

### Printer

Sovol Sv06

### Reproduction steps

1 I wanted to separate gcode file of each piece and made project for each gcode for each

2 Delete body to print head helmet backpack

3 moved the parts around to print efficiently lots of moves and rotates of parts
    saved project for that and gcode for that

4 then had to ctrl z probably many times maybe 15 times,to go back to astronaut body
    so i can then make separate project and gcode to save for only body

5 Cura give error code tttttttttthat slicing failed. 

6 First time this ever happened with Cura for me, and I have installed many various versions to try

7 Thank you soooo much for making the program and all the work team does, i alwasy use Cura, don't like Prusa

### Actual results

 error slice fail

### Expected results

 should slice and option to preview as well

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I use your directions to try and find the log file for the cura, and it is not there. I go through all files inside of it with 2 different curas installed and i even search with search box to find cura.log and i can't find it , where the heck is it ?
[ss_med_astronaut_body_rescaled_to_76.21_cura_project.zip](https://github.com/Ultimaker/Cura/files/11358040/ss_med_astronaut_body_rescaled_to_76.21_cura_project.zip)
[ss_med_astronaut_allparts_rescaled_to_76.21_cura_project.zip](https://github.com/Ultimaker/Cura/files/11358041/ss_med_astronaut_allparts_rescaled_to_76.21_cura_project.zip)
[med_astronaut.zip](https://github.com/Ultimaker/Cura/files/11358063/med_astronaut.zip)
[med-astronaut.zip](https://github.com/Ultimaker/Cura/files/11358064/med-astronaut.zip)

GregValiant, 2023-04-29:
Thanks for the report and files.
There is a bug that can keep models from slicing when they are in certain positions on the build plate.  If you move or rotate the model you can often fool the bug.
[image]
[image]

I will mark this as a duplicate.

JoelStolarski, 2023-04-30:
Thank you so much for looking at the model. I'm always working with multiple saves, and backing up of files, plus did extensive work on the head in the Blender to find and remove the non-manifold sections. I use a Cloud computing service and at random times, I can lose connection, or the server gets heavy traffic, and then will quit. Putting parts, and deleting, or doing silly things like moving parts around, then ctrl z so many times back created confusing workflow for Cura to solve or as you mention, that spot where body was triggered error. Body sliced fine, after closing Cura then pause before opening again. At least in small part, I uploaded the error for Cura team to look at. Would still like to find that log file. Strange Win 10 couldn't find it.

JoelStolarski, 2023-04-30:
thx , replied on github in comment section

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

JoelStolarski, 2024-04-29:
I downloaded the 5.7.0 and it is slicing very well.