Ultimaker/Cura #15386: slicing process interrupted

hphilibert, 2023-04-29:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

I tried many different settings, but each time, the slicing stopped before ending, without any reason.

### Actual results

interruption in slicing process

### Expected results

i expected the slicing process to end normaly, generating a .ufp file.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11359652/cura.log)
[UMS5_RuinedHut-part2 v1.zip](https://github.com/Ultimaker/Cura/files/11359662/UMS5_RuinedHut-part2.v1.zip)
[UMS5_RuinedHut-part2 v2.zip](https://github.com/Ultimaker/Cura/files/11359663/UMS5_RuinedHut-part2.v2.zip)

GregValiant, 2023-04-30:
Thanks for the report and the files.
The v1 file hung during slicing.  I switched to my single extruder Ender and it sliced with no problem.
The Cura team will take a look.

ericspivey, 2023-05-05:
Hey @hphilibert , try installing v4.13.1 -- I was having the same problem, and that solved it.

hphilibert, 2023-05-14:
> Hey @hphilibert , try installing v4.13.1 -- I was having the same problem, and that solved it.

Thank you for the advice. Seems to be working. v4.13.1 did the slicing process successfully. 
I'm going to try the printing next week end. I keep you informed.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic